benlamonica/homebridge-rasppi-gpio-garagedoor #3: Open & Close Instead of Toggle.

simonrb2000, 2016-04-06:
Is it possible to have separate pins for open and close instead of a toggle? 

Cheers

toshibochan, 2016-04-07:
Probably no because garage door use only on push button

benlamonica, 2016-04-07:
It wouldn't be very hard to split it up so that you configured 2 pins. If you submit a pull request, I'll integrate it in. I don't have a lot of time right now to work on this project.

simonrb2000, 2016-04-07:
Ah ok. My garage door currently is only on a toggle but does have the option of separate open and close which I can connect to different pins. 

I'll do the pull request. Thank you.

simonrb2000, 2016-04-09:
Before I issue the pull request the Pi running homebridge isn't the Pi connected to my garage door. Is there a way to add an address so it uses the GPIO ports on the Pi connected to the garage door? I doubt it is possible so I'll have to do it another way maybe.

toshibochan, 2016-04-09:
You make start scrip on /etc/unit.d/?

toshibochan, 2016-04-09:
Go to README.md and on the beginning show to add start script to use PI GPIO pin for control the door and sensor. You need relay to control the door.

toshibochan, 2016-04-09:
Oh you have toggle switch!?

toshibochan, 2016-04-09:
You need 1 for open and 1 for close and 1 for sensor?

simonrb2000, 2016-04-09:
My Garage door can do either a toggle or an open and close. For now I have it connected to a toggle and pin 4 on the GPIO which is connected to a relay. Problem is the Raspberry Pi connected to my garage door via the GPIO doesn't have homebridge running on it. Homebridge is running on a Raspberry Pi inside the house. If I export the pins via your script it will expose the pins on the house pi and not the garage pi connected to the garage door. Hope that makes sense

toshibochan, 2016-04-09:
you can install homebridge on the PI on the garage to and add on the app to

toshibochan, 2016-04-09:
I have 2 homebridge running in my house!

simonrb2000, 2016-04-09:
I didn't think of that. I suppose it's just like adding another accessory to HomeKit. Might give that a go! Thank you!! :-)

toshibochan, 2016-04-09:
Just change the address of the homebridge. CC:22:.....

simonrb2000, 2016-04-09:
Is that the only thing to Change? Won't the port it listens on conflict?

toshibochan, 2016-04-10:
Is the only thing I changed!

toshibochan, 2016-04-10:
this is my gpio setting:

#!/bin/sh
### BEGIN INIT INFO
# Provides:          garage-door-gpio
# Required-Start:    $remote_fs dbus
# Required-Stop:     $remote_fs dbus
# Should-Start:      $syslog
# Should-Stop:       $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Enable Garage Door GPIO
# Description:       Enable Garage Door GPIO
### END INIT INFO

PATH=/sbin:/bin:/usr/sbin:/usr/bin
DESC="Enable GPIO for the Garage Door Module"
NAME="garage-door-gpio"
SCRIPTNAME=/etc/init.d/$NAME

. /lib/lsb/init-functions

case "$1" in
    start)
        log_daemon_msg "Enabling Garage Door GPIO"
        success=0
        if [ ! -e /sys/class/gpio/gpio23 ] ; then
                echo 23 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio23/direction
                echo 1 > /sys/class/gpio/gpio23/active_low
                echo 0 > /sys/class/gpio/gpio23/value
        fi

        if [ ! -e /sys/class/gpio/gpio17 ] ; then
                echo 17 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio17/direction
                echo 1 > /sys/class/gpio/gpio17/active_low
                echo 0 > /sys/class/gpio/gpio17/value
        fi

        if [ ! -e /sys/class/gpio/gpio24 ] ; then
                echo 24 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio24/direction
        fi

        if [ ! -e /sys/class/gpio/gpio27 ] ; then
                echo 27 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio27/direction
        fi
 
        log_end_msg $success
        ;;
    *)
        echo "Usage: $SCRIPTNAME {start}" >&2
        exit 1
        ;;
esac

exit 0

toshibochan, 2016-04-10:
this is my GPIO setting:

        if [ ! -e /sys/class/gpio/gpio23 ] ; then
                echo 23 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio23/direction
                echo 1 > /sys/class/gpio/gpio23/active_low
                echo 0 > /sys/class/gpio/gpio23/value
        fi

        if [ ! -e /sys/class/gpio/gpio17 ] ; then
                echo 17 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio17/direction
                echo 1 > /sys/class/gpio/gpio17/active_low
                echo 0 > /sys/class/gpio/gpio17/value
        fi

        if [ ! -e /sys/class/gpio/gpio24 ] ; then
                echo 24 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio24/direction
        fi

        if [ ! -e /sys/class/gpio/gpio27 ] ; then
                echo 27 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio27/direction
        fi

toshibochan, 2016-04-10:
for homebridge start on boot i use this site example:
http://www.kramnameloc.com/homebridge-pi-switch

simonrb2000, 2016-04-10:
Just tying to get my head around it. When you close the garage door it will check the state pin (I have a switch on the floor connected to pi pin 14 so I know when it's closed) to make sure it closes and if it doesn't close by a certain time, does it try to close again as its a toggle switch?

toshibochan, 2016-04-10:
I don't think will try to close automatically.

simonrb2000, 2016-04-13:
I can only get this half working. I have managed to get the door status however whenever I try to open the door Homebridge crashes. I have exposed the pins and they show up. I am using Domoticz on the same Pi Homebridge and it can open the garage door by operating the pin. Seems to be it cannot write to the pin. Please see below - thank you!

```
[4/13/2016, 1:52:40 AM] [Garage Door] Door Sensor Pin: 14
[4/13/2016, 1:52:40 AM] [Garage Door] Door Poll in ms: 4000
[4/13/2016, 1:52:40 AM] [Garage Door] Door Opens in seconds: 14
[4/13/2016, 1:52:40 AM] [Garage Door] Setting state to 1
Scan this code with your HomeKit App on your iOS device to pair with Homebridge:
                       
    ┌────────────┐     
    │ 031-45-154 │     
    └────────────┘     
                       
[4/13/2016, 1:52:40 AM] Homebridge is running on port 51929.
[4/13/2016, 1:53:08 AM] [Garage Door] GarageDoor is CLOSED (1)
[4/13/2016, 1:53:16 AM] [Garage Door] Setting state to 0
[4/13/2016, 1:53:16 AM] [Garage Door] Triggering GarageDoor Relay
fs.js:549
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: EACCES: permission denied, open '/sys/class/gpio/gpio4/value'
    at Error (native)
    at Object.fs.openSync (fs.js:549:18)
    at Object.fs.writeFileSync (fs.js:1155:15)
    at Object.RaspPiGPIOGarageDoorAccessory.setState (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:103:12)
    at emitThree (events.js:97:13)
    at emit (events.js:175:7)
    at Characteristic.setValue (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Characteristic.js:155:10)
    at Bridge.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:709:22)
    at Array.forEach (native)
    at Bridge.Accessory._handleSetCharacteristics (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:654:8)
pi@raspberrypi ~ $ 
```

benlamonica, 2016-04-13:
Are you running homebridge as root? And can you verify that the file exists?

--Ben L

simonrb2000, 2016-04-13:
Not running it as root as I’d rather not. When I do run it as root I can’t even get it to show up in homekit. What file? I have domoticz running on the Pi and that has access to pin 4 and can open the garage door so not sure why homebridge can’t access it.

benlamonica, 2016-04-13:
GPIO access requires something to run as root. If you don't want homebridge to run as root, you will need to make a helper that does. Or you can figure out how domoticz is doing it. 

--Ben L

simonrb2000, 2016-04-13:
I would run Domoticz as root however it doesn’t work when I run it as root. When I run it normally I get the:-

> [4/13/2016, 1:52:40 AM] [Garage Door] Door Sensor Pin: 14

If I run it as root I don’t, it loads up but listens on a different port also. Can’t add it to eve and the normal home kit can’t see the garage door when ran as root. Is there another config file for root? To load homebridge I type homebridge at the command prompt and sudo homebridge for root. Is this right? 

Cheers

simonrb2000, 2016-04-13:
Did you mean the file '/sys/class/gpio/gpio4/value’?  As thats empty. Its empty after gpio

simonrb2000, 2016-04-13:
I have WiringPi installed and I think that's how Domoticz accesses the GPIO. I have the pin exposed as is shows up as exposed as output and domoticz can open the garage door yet when I try and say Siri open garage door, homebridge crashes. Yet I can ask if it's open and it reads the pin state correctly if it's open or closed. Strange.

simonrb2000, 2016-04-13:
Hours on it last night, till 3 am and got it working within minutes this morning before work. Domoticz was being it as Pin 4 for and GPIO sees it as pin 7 even though its pin 4. Just exported pin 7 and all is working great. Even opening it from garage remote triggers a notification its open, Thanks a million! Onto the next project hehe

benlamonica, 2016-05-14:
Glad you got it to work. Closing issue.

closed 2016-05-14 by benlamonica