benlamonica/homebridge-rasppi-gpio-garagedoor #4: Help with issue

Azza1979, 2016-05-19:
Hi
Thank you for providing this platform, I'm having issues with the installation of it but that could be just me with very limited coding knowledge. I have homebridge running on Rpi 3 and using the wemo platform with nil issues but ive been trying to follow your instructions on this garage door but just when think i have it right i now have this error. Any assistance would be great

Cheers Aaron 

pi@raspberrypi:~ $ homebridge
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
[5/19/2016, 9:48:57 PM] Loaded plugin: homebridge-platform-wemo
[5/19/2016, 9:48:57 PM] Registering platform 'homebridge-platform-wemo.BelkinWeMo'
[5/19/2016, 9:48:57 PM] ---
[5/19/2016, 9:48:57 PM] Loaded plugin: homebridge-rasppi-gpio-garagedoor
[5/19/2016, 9:48:57 PM] Registering accessory 'homebridge-rasppi-gpio-garagedoor.RaspPiGPIOGarageDoor'
[5/19/2016, 9:48:57 PM] ---
[5/19/2016, 9:48:57 PM] Loaded config.json with 1 accessories and 1 platforms.
[5/19/2016, 9:48:57 PM] ---
[5/19/2016, 9:48:57 PM] Loading 1 platforms...
[5/19/2016, 9:48:57 PM] [WeMo Platform] Initializing BelkinWeMo platform...
[5/19/2016, 9:48:57 PM] [WeMo Platform] Wemo Platform Plugin Loaded 
[5/19/2016, 9:48:57 PM] [WeMo Platform] Fetching the Wemo Accessories, expecting 20 and will wait 25 seconds to find them.
[5/19/2016, 9:48:57 PM] Loading 1 accessories...
[5/19/2016, 9:48:57 PM] [Front Garage Door] Initializing RaspPiGPIOGarageDoor accessory...
[5/19/2016, 9:48:57 PM] [Front Garage Door] Door Switch Pin: 17
[5/19/2016, 9:48:57 PM] [Front Garage Door] Door Sensor Pin: 27
[5/19/2016, 9:48:57 PM] [Front Garage Door] Door Poll in ms: 4000
[5/19/2016, 9:48:57 PM] [Front Garage Door] Door Opens in seconds: 14
fs.js:585
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open '/sys/class/gpio/gpio27/value'
    at Error (native)
    at Object.fs.openSync (fs.js:585:18)
    at Object.fs.readFileSync (fs.js:432:33)
    at Object.RaspPiGPIOGarageDoorAccessory.isClosed (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:70:15)
    at Object.RaspPiGPIOGarageDoorAccessory.initService (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:51:25)
    at Object.RaspPiGPIOGarageDoorAccessory (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:23:8)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/lib/server.js:244:29)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:72:38)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:40:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
pi@raspberrypi:~ $

simonrb2000, 2016-05-19:
Does pin 27 have a value? Mine is 1 is closed and 0 is open.

at Object. (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
pi@raspberrypi:~ $

-

Azza1979, 2016-05-19:
Gpio 27 is for my open or closed state. I just used the script that was provided just changed the Gpio

toshibochan, 2016-05-19:
if [ ! -e /sys/class/gpio/gpio23 ] ; then
            echo 23 > /sys/class/gpio/export
            success=$?
            echo out > /sys/class/gpio/gpio23/direction
            echo 1 > /sys/class/gpio/gpio23/active_low
            echo 0 > /sys/class/gpio/gpio23/value
    fi

    if [ ! -e /sys/class/gpio/gpio17 ] ; then
            echo 17 > /sys/class/gpio/export
            success=$?
            echo out > /sys/class/gpio/gpio17/direction
            echo 1 > /sys/class/gpio/gpio17/active_low
            echo 0 > /sys/class/gpio/gpio17/value
    fi

    if [ ! -e /sys/class/gpio/gpio24 ] ; then
            echo 24 > /sys/class/gpio/export
            success=$?
            echo in > /sys/class/gpio/gpio24/direction
    fi

    if [ ! -e /sys/class/gpio/gpio27 ] ; then
            echo 27 > /sys/class/gpio/export
            success=$?
            echo in > /sys/class/gpio/gpio27/direction
    fi

toshibochan, 2016-05-19:
Because I have 2 door
23 and 17 is for the door opener 
24 and 27 is for the door sensor

Azza1979, 2016-05-19:
Thanks for the input I will look at that in the morning my good ol internet failed.

toshibochan, 2016-05-19:
You have to connect raspberry pi to relay and relay to door opener

Azza1979, 2016-05-19:
So i followed the that script and edited mine but i still get the same issues. Would it be a permissions error

Azza1979, 2016-05-19:
Ok so i have managed to get it going but he state reads open all the time. I have 1 wire from the sensor connected to gpio 24  and the other to ground is this right if so what im i doing wrong

closed 2016-05-19 by Azza1979

toshibochan, 2016-05-19:
The light on the relay is on or off?

toshibochan, 2016-05-19:
This pin is for sensor? Look like the developer did to work with 3.3v and pin 24 if is sensor

Azza1979, 2016-05-19:
All good mate I changed the second wire to the 3.3. Would you mind sending me your JSON file and script as I have a second door to add as well thanks.

toshibochan, 2016-05-20:
I'm not in town now! Be back home in 2 weeks!  But just add more accessories. I think my was:

   "accessories": [{
            "accessory": "RaspPiGPIOGarageDoor",
            "name": "Garage Door",
            "doorSwitchPin": 23,
            "doorSensorPin": 24,
            "doorPollInMs": 4000,
            "doorOpensInSeconds": 14
        },{
            "accessory": "RaspPiGPIOGarageDoor",
            "name": "Garage Door",
            "doorSwitchPin": 17,
            "doorSensorPin": 27,
            "doorPollInMs": 4000,
            "doorOpensInSeconds": 14
        }
    ],

toshibochan, 2016-05-20:
On the "name" change the name you want: for example;
"name":"Garage Door",

"name":"Gate",

toshibochan, 2016-05-20:
Past your config.json here, I can take look

Azza1979, 2016-05-20:
Ok so i have managed to get both garages working but when i connect the second garage sensor to the second 3.3v it trips the rpi. Should i have resistors on the sensor

toshibochan, 2016-05-20:
I ask the developed but he sad toconnect direct! I don't know what is correct!

simonrb2000, 2016-05-20:
Think that depends on sensor. I used a switch but I put a resistor on the earth wire. That's just to stop the floating voltage giving you a false reading. Without it though it shouldn't trip the Pi.

toshibochan, 2016-05-20:
How mutch is you resistor?

simonrb2000, 2016-05-20:
Can't remember now. Was very small. 20k I think

toshibochan, 2016-05-21:
You think 420ohm or 220ohm will work?

benlamonica, 2016-05-21:
It should be at least 10k ohms. You don't want very much current to flow through it.