benlamonica/homebridge-rasppi-gpio-garagedoor #5: doorSensorPin-less

mattgraham, 2016-06-28:
My current setup doesn't have a doorSensorPin and it appears that the script isn't happy with me because of it. Do I run a magnetic switch to overcome this or is there a better way around.    

Secondly, running into what I think is a 'root' issue with the GPIO. The relay I've confirmed works when I

```
pi@homebridge:~ $ gpio write 7 1
pi@homebridge:~ $ gpio mode 7 out 
pi@homebridge:~ $ gpio mode 7 in
```

Any suggestions to what I'm doing wrong? Cheers

mattgraham, 2016-06-28:
For what its worth; this was the last guide I had running, however, without homebridge (which is a must): http://www.instructables.com/id/Raspberry-Pi-Garage-Door-Opener/step6/Create-a-Startup-Service/

toshibochan, 2016-06-30:
hi @mattgraham, if you no have the door magnet sensor the homebridge think is open and never close. better to add the sensor.

mattgraham, 2016-06-30:
Then maybe a noob question; magnet sensors have two wires; where does the second go? Back to another ground?

toshibochan, 2016-06-30:
this is the file for my gate and garage switch and the sensor. in my case the 23 and 17 is switch and 24 and 27 is for sensor.
@benlamonica told me one go to 3v and other go to sensor pin you chose. in my case 3v and 24 and 27
for the switch go to relay.
[garage-door-gpio.txt](https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor/files/342469/garage-door-gpio.txt)

benlamonica, 2016-11-29:
I've made it so that you can not configure any sensor pins, and it will just rely on it's internal state. Please install version 1.0.3 and test. If there is still a problem, please open a new issue.

benlamonica, 2016-11-29:
Or re-open this one.

closed 2016-11-29 by benlamonica