benlamonica/homebridge-rasppi-gpio-garagedoor #8: change switching value of relay / make it configurable

michis0806, 2016-09-05:
Hi benlamonica,

I like your module very much. But I'd like to change the switching value of the relay-switch. With the current settings, the relay switches when it's powerless, so when the rasppi reboots, my garage door opens. 

I switched the value in #74 to 1 and in #103 to 0 in the configuration. so the door stays closed when the system is powerless...

Thanks!
Michael

muhackl, 2016-10-10:
Hi @michis0806 and @benlamonica ,
I'm also interested in what @michis0806 asked. I also would be happy, if I could edit the time the relay switches. At the moment, it's 1000 ms. How can I edit it, after I installed the plugin?
Many thanks and best regards
Muhackl

benlamonica, 2016-10-10:
I will make the switch time configurable and also if the should be high or low to trigger it. I'll submit the new build sometime today. 

--Ben L

muhackl, 2016-10-10:
Sounds great. Really looking forward to it and appreciate your work!

muhackl, 2016-10-12:
@benlamonica Did you have the chance to work on the switch and the trigger? Really looking forward to it.

sogseal, 2016-11-17:
@benlamonica I Appriciate your work, got my garage door all setup and wife is happy! I have two questions. When i ask siri The status of the garage door, when its closed/open she replies closing/opening is there away to fix that(p.s on home app everything looks good). And last one, could you possibly add another sensor on the opposite side, like it triggers then the door is open, so if the door got stopped in between to sensors it would tell you opening/closing/partially opened.

fiddypal, 2016-11-23:
Guys i was running into the same problem and found the 1/0's i needed to flip in the index.js file. I think i see how the variables are getting pulled from the config.json (i have them hardcoded right now) so let me play around with it over the holidays this week and see if i can get it working with insert variables in the config.json.

fiddypal, 2016-11-23:
OK i was able to figure it out.

Use this file for index.js in the plugin folder:

```
var fs = require("fs");
var Service, Characteristic, DoorState; // set in the module.exports, from homebridge

module.exports = function(homebridge) {
  Service = homebridge.hap.Service;
  Characteristic = homebridge.hap.Characteristic;
  DoorState = homebridge.hap.Characteristic.CurrentDoorState;

  homebridge.registerAccessory("homebridge-rasppi-gpio-garagedoor", "RaspPiGPIOGarageDoor", RaspPiGPIOGarageDoorAccessory);
}

function RaspPiGPIOGarageDoorAccessory(log, config) {
  this.log = log;
  this.name = config["name"];
  this.defaultSensorValue = config["defaultSensorValue"];
  this.relayDefaultValue = config["relayDefaultValue"];
  this.relaySwitchValue = config["relaySwitchValue"];
  this.doorSwitchPin = config["doorSwitchPin"];
  this.doorSensorPin = config["doorSensorPin"];
  this.doorPollInMs = config["doorPollInMs"];
  this.doorOpensInSeconds = config["doorOpensInSeconds"];
  log("Door Switch Pin: " + this.doorSwitchPin);
  log("Door Sensor Pin: " + this.doorSensorPin);
  log("Door Poll in ms: " + this.doorPollInMs);
  log("Door Opens in seconds: " + this.doorOpensInSeconds);
  this.initService();
  setTimeout(this.monitorDoorState.bind(this), this.doorPollInMs);
}

RaspPiGPIOGarageDoorAccessory.prototype = {

  monitorDoorState: function() {
     var isClosed = this.isClosed();
     if (isClosed != this.wasClosed) {
       this.wasClosed = isClosed;
       var state = isClosed ? DoorState.CLOSED : DoorState.OPEN;       
       this.log("Door state changed to " + (isClosed ? "CLOSED" : "OPEN"));
       if (!this.operating) {
         this.currentDoorState.setValue(state);
         this.targetDoorState.setValue(state);
         this.targetState = state;
       }
     }
     setTimeout(this.monitorDoorState.bind(this), this.doorPollInMs);
  },

  initService: function() {
    this.garageDoorOpener = new Service.GarageDoorOpener(this.name,this.name);
    this.currentDoorState = this.garageDoorOpener.getCharacteristic(DoorState);
    this.currentDoorState.on('get', this.getState.bind(this));
    this.targetDoorState = this.garageDoorOpener.getCharacteristic(Characteristic.TargetDoorState);
    this.targetDoorState.on('set', this.setState.bind(this));
    this.targetDoorState.on('get', this.getTargetState.bind(this));
    var isClosed = this.isClosed();
    this.currentDoorState.setValue(isClosed ? DoorState.CLOSED : DoorState.OPEN);
    this.targetDoorState.setValue(isClosed ? DoorState.CLOSED : DoorState.OPEN);
    this.infoService = new Service.AccessoryInformation();
    this.infoService
      .setCharacteristic(Characteristic.Manufacturer, "Opensource Community")
      .setCharacteristic(Characteristic.Model, "RaspPi GPIO GarageDoor")
      .setCharacteristic(Characteristic.SerialNumber, "Version 1.0.0");
  
    this.wasClosed = isClosed;
    this.operating = false;
    setTimeout(this.monitorDoorState.bind(this), this.doorPollInMs);
  },

  getTargetState: function(callback) {
    callback(null, this.targetState);
  },

  isClosed: function() {
    return fs.readFileSync("/sys/class/gpio/gpio"+this.doorSensorPin+"/value", "utf8").trim() == this.defaultSensorValue;
  },

  switchOff: function() {
    fs.writeFileSync("/sys/class/gpio/gpio"+this.doorSwitchPin+"/value", this.relayDefaultValue);
    this.log("Turning off GarageDoor Relay");
  },

  setFinalDoorState: function() {
    var isClosed = this.isClosed();
    if ((this.targetState == DoorState.CLOSED && !isClosed) || (this.targetState == DoorState.OPEN && isClosed)) {
      this.log("Was trying to " + (this.targetState == DoorState.CLOSED ? " CLOSE " : " OPEN ") + "the door, but it is still " + (isClosed ? "CLOSED":"OPEN"));
      this.currentDoorState.setValue(DoorState.STOPPED);
      this.targetDoorState.setValue(isClosed ? DoorState.CLOSED : DoorState.OPEN);
    } else {
      this.currentDoorState.setValue(this.targetState);
    }
    this.operating = false;
  },

  setState: function(state, callback) {
    this.log("Setting state to " + state);
    this.targetState = state;
    var isClosed = this.isClosed();
    if ((state == DoorState.OPEN && isClosed) || (state == DoorState.CLOSED && !isClosed)) {
        this.log("Triggering GarageDoor Relay");
        this.operating = true; 
        if (state == DoorState.OPEN) {
            this.currentDoorState.setValue(DoorState.OPENING);
        } else {
            this.currentDoorState.setValue(DoorState.CLOSING);
        }
        setTimeout(this.setFinalDoorState.bind(this), this.doorOpensInSeconds * 1000);
        fs.writeFileSync("/sys/class/gpio/gpio"+this.doorSwitchPin+"/value", this.relaySwitchValue);
        setTimeout(this.switchOff.bind(this), 1000);
    }

    callback();
    return true;
  },

  getState: function(callback) {
    var isClosed = this.isClosed();
    this.log("GarageDoor is " + (isClosed ? "CLOSED ("+DoorState.CLOSED+")" : "OPEN ("+DoorState.OPEN+")")); 
    callback(null, (isClosed ? DoorState.CLOSED : DoorState.OPEN));
  },

  getServices: function() {
    return [this.infoService, this.garageDoorOpener];
  }
};

```

Here is example how to setup your config.json in homebridge:

```
        "accessories": [{
                "accessory": "RaspPiGPIOGarageDoor",
                "name": "Garage Door",
                "defaultSensorValue": "0",
                "relayDefaultValue": "1",
                "relaySwitchValue": "0",
                "doorSwitchPin": 7,
                "doorSensorPin": 18,
                "doorPollInMs": 4000,
                "doorOpensInSeconds": 14
        }, 
```

deafultSensorValue - Set this to 0 if using an NC sensor, 1 if using NO sensor.
relayDefaultValue - Set to 1 if your relay is jamming in the "ON" position when not in use (you also need to set the relaySwitchValue to the invese of whatever this value is)
relaySwitchValue - Set to inverse of relayDefaultValue

benlamonica, 2016-11-26:
@sogseal - have you upgraded to iOS 10 and the latest version of homebridge? When I just recently upgraded it fixed the status messages.

simonrb2000, 2016-11-26:
I have the same as above. Been using this for over a year and I used to show people asking Siri if my garage door is open? She would reply closed. However past few weeks she has been saying closing when its closed and opening when its open. I have latest iOS 10 and latest Homebridge.

sogseal, 2016-11-26:
@sogseal, i have latest ios and installed homebridge a week ago.

benlamonica, 2016-11-26:
I’ll take a look at it.

simonrb2000, 2016-11-26:
It's only Siri who gets the status wrong. The Home App shows it as opening when it's opening and as open once it's open. Same as the closing as it closes and closed once it's closed. Only Siri will say it's closing when asked even though it's closed. (Same with opening) hope thats a bit clearer :-)

fiddypal, 2016-11-26:
Same problem here. Status works correctly in the home app, but when you ask Siri 'is my garage open ' she will respond with 'your garage is opening ' when it's open and 'your garage is closing' when it's closed. Never the correct open or closed status. Had the same problem when running straight off hap-nodejs and other plugins as well.

fiddypal, 2016-11-26:
@simonrb2000 

My fix is for a couple things. One was that I am using an NC contact switch, the stock code is hard coded for an NO contact switch. This was causing incorrect open and close reading.

The other problem I had was the relay was jamming in the on position at rest. This would only allow you to use the app to open and close, it blocked out the switch on the wall.

So instead of hard coding the opposite values to fix these issues I pulled them out into the config to make it a lot easier to toggle for people.

benlamonica, 2016-11-27:
I’ve spent a lot of time implementing everything everyone has asked for today.

Configurable pins, an open door sensor (though I can’t test that one), configurable time to press the button, and I tried to fix the bug with Siri, but I think it’s an apple bug, because when you query Siri, you see in the logs that homebridge is reporting CLOSED to siri, but then siri turns around and says Closing.

I tried migrating to wiring-pi in order to have rootless homebridge, but couldn’t get it to work in nodejs (works fine from command line…). So I switched back to the filesystem based pin triggering. I did add an error if you try to start without being root so that at least you know it’s needed up front.

I’ll get these changes published either tonight or on Monday.

simonrb2000, 2016-11-27:
Appreciate all your hard work mate, we all do. It had been working fine, Siri saying it's closed when it was closed. It's only been recently it's changed and I hadn't changed anything in my setup so I suspect your right, it's an Apple thing.

Thanks for looking :-)

-

sogseal, 2016-11-27:
@benlamonica you are my hero man! love what you did, using it everyday, my pregnant wife is happy as well(happy wife, happy life). Now i need to buy apple TV to control home away.

fiddypal, 2016-11-27:
Ben I also want to thank you for this awesome software. It's been an awesome and very rewarding project for learning Linux and some basic coding skills. 

@sogseal

I believe the obtrusion detection input is for the little 'eyes' at the bottom of the door that form an invisible trip wire to stop the door from closing if you are walking under it. I have not wired this up yet but plan to in the future, I'm sure it's a simple job garage door wiring is not very complex.

sogseal, 2016-11-27:
@fiddypal yes, i was wondering how to implemet those 'eyes' to a pi and the config, i have no js knowlegde, but a little python. this is all so much fun!

fiddypal, 2016-11-27:
It should be as simple as copying the sensor detection code as used by the open/close sensor and just have it trigger the obstruction detected characteristic to true when your garage 'eyes' sees an obstruction. My guess is this has embedded functionality in the HomeKit framework to update the door Status to 'jammed' or such when you set the flag to true. Let me go hook up multi meter to the eyes and see what kind of signal I get back when I block them.

https://developer.apple.com/reference/homekit/hmcharacteristictypeobstructiondetected

benlamonica, 2016-11-27:
The software part should be pretty easy. I think intercepting the sensor will be more difficult, as you’ll have to tie the two electrical systems together. If you want to give it a try on the software side, you can code it and submit a pull request. I’d be happy to offer advice too.

benlamonica, 2016-11-27:
I've committed the changes, so if you want to make changes, please pull down master before you do, as I've done some refactoring.

fiddypal, 2016-11-27:
@benlamonica 
You are right, after a quick google it seems the sensors are sending a looping signal back to the garage unit when unblocked, then when you block them one wire is 6v above the other, so I'd imagine some intermediate hardware would be required as it can't be wired straight to the gpio pins on the pi. Unfortuntely I'm not that skilled with hardware yet but hopefully someone can chime in and move us along, I can certainly help on the software side.

sogseal, 2016-11-27:
ill check on the hw, ill play with my garage door 'eyes' to get my calculations, from there ill see if i have supplies to add them to pi/gpio's.

sogseal, 2016-11-27:
so pi dont have Analog pins if im not mistaken. i check out the 'eyes', when blocked i get  6.150V and when no obstructions i get 5.840V, very fine difference. Prolly need some kind of analog converter so the pi can read voltage.

fiddypal, 2016-11-27:
@benlamonica can you please share example of new config.json since you have updated the code base.

Thanks!

benlamonica, 2016-11-27:
Good point. Forgot to update the docs. Will do that tomorrow. 

--Ben L

benlamonica, 2016-11-29:
I've released version 1.0.3 of the plugin. This makes all of the pins configurable to be active_low or active_high. Please see the readme. Closing this Issue. Please open a new issue if you have other problems.

closed 2016-11-29 by benlamonica