benlamonica/homebridge-rasppi-gpio-garagedoor #9: Crash when triggering relay

schweppy, 2016-10-10:
Hi

Bit of a noob but I am getting the following error when trying to operate the relay. Any idea as to what is wrong?

[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] Loaded plugin: homebridge-rasppi-gpio-garagedoor
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] Registering accessory 'homebridge-rasppi-gpio-garagedoor.RaspPiGPIOGarageDoor'
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] ---
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] Loaded config.json with 1 accessories and 0 platforms.
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] ---
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] Loading 1 accessories...
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] [Garage Door] Initializing RaspPiGPIOGarageDoor accessory...
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] [Garage Door] Door Switch Pin: 26
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] [Garage Door] Door Sensor Pin: 1
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] [Garage Door] Door Poll in ms: 4000
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] [Garage Door] Door Opens in seconds: 20
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] [Garage Door] Setting state to 1
Scan this code with your HomeKit App on your iOS device to pair with Homebridge:
[Mon Oct 10 2016 12:38:00 GMT+0000 (UTC)] Homebridge is running on port 51826.
[Mon Oct 10 2016 12:38:26 GMT+0000 (UTC)] [Garage Door] GarageDoor is CLOSED (1)
[Mon Oct 10 2016 12:38:32 GMT+0000 (UTC)] [Garage Door] Setting state to 0
[Mon Oct 10 2016 12:38:32 GMT+0000 (UTC)] [Garage Door] Triggering GarageDoor Relay
fs.js:663
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EPERM: operation not permitted, write
    at Error (native)
    at Object.fs.writeSync (fs.js:663:20)
    at Object.fs.writeFileSync (fs.js:1170:24)
    at Object.RaspPiGPIOGarageDoorAccessory.setState (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:103:12)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at Characteristic.setValue (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Characteristic.js:155:10)
    at Bridge.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:727:22)
    at Array.forEach (native)
    at Bridge.Accessory._handleSetCharacteristics (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:672:8)

benlamonica, 2016-10-10:
> at Bridge. (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:727:22)

Are you running homebridge as root? Writing to the gpio pins requires root access. 

--Ben L

fiddypal, 2016-11-23:
I am having the same issues if i try and let things run from autostart. I followed the directions exactly, and the services run at startup but not in root because if i stop them and run again, i get the crash error that they cant write.

If i stop all services, then do sudo su, sudo homebridge, it works fine.

Can someone please explain to us linux noobs how to ensure these scripts are being setup correctly to run as root on startup?

I am using a raspberry pi with raspbian jesse.

My understanding is that by doing the CHMOD on the scripts you make in the init.d director, that makes them run as root? Or am i way off base here? does not seem to be working for me...

benlamonica, 2016-11-25:
@fiddypal - the start script that I provided should automatically run as root at startup. Did you run this after you copied the start script over?

sudo update-rc.d /etc/init.d/homebridge defaults

fiddypal, 2016-11-28:
I just did a fresh install and am having the same issues. The garage-door-gpio service boots and runs fine, but for some reason Homebridge will not. When I look at the logs it appears the service is trying to start over and over again. If I stop them both, boot into screen then run the garage-door-gpio service and run homebridge from command line (not as service) it works ! Don't know what I am doing wrong following the instructions  exactly!

fiddypal, 2016-11-28:
So for anyone having the same issues as me, here is what i have done to remedy the situation.

1. Still setup the "garage-door-gpio" script to run from /etc/init.d/ as described in the directions. 
(i dont know how to translate this over to systemd or booting as a service in screen so hands are kinda tied here, ideally i'd like to change this to boot from screen as a servivce but this setup works.)

2. install "screen" (this basically lets you keep a terminal session open in the background when SSH is closed, or like i am doing you can have your PI boot up a screen instance and run homebridge as root in there. Very handy program!)
```
sudo apt-get install screen
```

3. modify your /etc/rc.local file to include the following before the "exit 0" line at the end:
```
sudo su

su -c "screen -dmS homebridge homebridge" -s /bin/sh root
```

Now your PI should boot up with a screen session named "homebridge" that runs the homebridge service as root.

If you want to jump in and see whats going on, just enter:
```
screen -r
```
which will resume your screen session (if you have multiple screens open may need to specify which ones to resume).

fiddypal, 2016-11-29:
@benlamonica 
my success was short lived, now getting error which i know has to do with the script not running as root but i have followed all the directions :( :

```
[11/28/2016, 11:57:56 PM] [Garage Door] GarageDoor is CLOSED (1)
[11/28/2016, 11:57:56 PM] [LiFx] Living Room Light 1 - Get power: 1
[11/28/2016, 11:57:56 PM] [LiFx] Living room light 2 - Get power: 1
[11/28/2016, 11:57:58 PM] [Garage Door] Setting state to 0
[11/28/2016, 11:57:58 PM] [Garage Door] Triggering GarageDoor Relay
fs.js:710
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EPERM: operation not permitted, write
    at Object.fs.writeSync (fs.js:710:20)
    at Object.fs.writeFileSync (fs.js:1224:24)
    at RaspPiGPIOGarageDoorAccessory.setState (/usr/local/lib/node_modules/homeb                                                                                        ridge-rasppi-gpio-garagedoor/index.js:106:12)
    at emitMany (events.js:127:13)
    at Characteristic.TargetDoorState.emit (events.js:201:7)
    at Characteristic.TargetDoorState.Characteristic.setValue (/usr/local/lib/no                                                                                        de_modules/homebridge/node_modules/hap-nodejs/lib/Characteristic.js:155:10)
    at Bridge.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/h                                                                                        ap-nodejs/lib/Accessory.js:727:22)
    at Array.forEach (native)
    at Bridge.Accessory._handleSetCharacteristics (/usr/local/lib/node_modules/h                                                                                        omebridge/node_modules/hap-nodejs/lib/Accessory.js:672:8)
    at emitMany (events.js:127:13)

```

benlamonica, 2016-11-29:
please execute the following from the commandline:
ps aux | grep homebridge
uname -a
cat /etc/os-release
node --version
homebridge --version

And paste the results to this thread.

benlamonica, 2016-11-29:
Also, please upgrade to version 1.0.5.

fiddypal, 2016-11-29:
ps aux | grep homebridge
```
Last login: Tue Nov 29 00:24:21 2016 from iphone
pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# ps aux | grep homebridge
root       585  0.0  1.2   5192  2288 ?        Ss   00:25   0:00 SCREEN -dmS hom                ebridge homebridge
root       587  0.3 23.0 155492 42820 pts/1    Ssl+ 00:25   2:37 homebridge                     
root       976  0.0  1.0   4268  1908 pts/0    S+   12:19   0:00 grep homebridge
```

uname -a
```
Linux raspberrypi 4.4.32+ #924 Tue Nov 15 18:06:42 GMT 2016 armv6l GNU/Linux
```

cat /etc/os-release
```

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

node --version
```
v7.1.0
```

homebridge --version
```
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
0.4.10
```

ls -al /sys/class/gpio/gpio*/
```
/sys/class/gpio/gpio18/:
total 0
drwxrwx--- 3 root gpio    0 Nov 29 00:24 .
drwxr-xr-x 5 root root    0 Nov 29 00:24 ..
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 active_low
lrwxrwxrwx 1 root gpio    0 Nov 29 00:24 device -> ../../../20200000.gpio
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 direction
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 edge
drwxrwx--- 2 root gpio    0 Nov 29 00:24 power
lrwxrwxrwx 1 root gpio    0 Nov 29 00:24 subsystem -> ../../../../../../class/gpio
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 uevent
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 value

/sys/class/gpio/gpio7/:
total 0
drwxrwx--- 3 root gpio    0 Nov 29 00:24 .
drwxr-xr-x 5 root root    0 Nov 29 12:23 ..
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 active_low
lrwxrwxrwx 1 root gpio    0 Nov 29 00:24 device -> ../../../20200000.gpio
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 direction
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 edge
drwxrwx--- 2 root gpio    0 Nov 29 00:24 power
lrwxrwxrwx 1 root gpio    0 Nov 29 00:24 subsystem -> ../../../../../../class/gpio
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 uevent
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 value

/sys/class/gpio/gpiochip0/:
total 0
drwxrwx--- 3 root gpio    0 Nov 29 00:24 .
drwxr-xr-x 5 root root    0 Nov 29 12:23 ..
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 base
lrwxrwxrwx 1 root gpio    0 Nov 29 00:24 device -> ../../../20200000.gpio
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 label
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 ngpio
drwxrwx--- 2 root gpio    0 Nov 29 00:24 power
lrwxrwxrwx 1 root gpio    0 Nov 29 00:24 subsystem -> ../../../../../../class/gpio
-rwxrwx--- 1 root gpio 4096 Nov 29 00:24 uevent
```

fiddypal, 2016-11-29:
@benlamonica Just updated to 1.0.5 as well and same error when i try to open the garage.

fiddypal, 2016-11-29:
@benlamonica - I figured out the problem and its a total noob mistake... I was setting my sensor pin to "OUT" in the init.d script, and my relay pin to "IN". So when i tried to open the door, it was trying to execute a write against a pin set to IN, not OUT.

Swapped those around in the boot-up script and now i am golden! Sorry for wasting your time with such a  noob mistake, appreciate the help.

For those wondering how i figure this out, the following error message played a key role:
```
Error: EPERM:
```
EPERM means there was a logical error, like trying to write to a pin set as IN, passing a bad value, etc, not a permissions error.

benlamonica, 2016-11-29:
Wonderful! Glad you figured it out. Enjoy your automated garage door :)

--Ben L

closed 2016-11-29 by benlamonica