benlamonica/homebridge-rasppi-gpio-garagedoor #12: Relay locking in on position upon first boot

fiddypal, 2016-11-29:
Weird issues here after updating to the new 1.0.5 code.

1 thing i notice is that "doorSwitchValue" in the config.json is not referenced in the index.js file, i believe this should replace "relayOnValue" on line 38.

My relay values are flipped, so i had to add that config parameter to the config.json, which leads me to my next issue of the relay being "stuck" in the on position upon boot.

When i fire up homebridge, the relay triggers to on and my garage opens, and the relay stays jammed in the on position. (this prevents me from using the button on the wall)

Once i press the button to close the garage, it then does the proper cycle of "press button for 1 sec" then shut the relay back off. (or whatever time you defined in "doorSwitchPressTimeInMs"). After i press the button once in the app it seems to THEN set the relay to off in its default state.

Is there a way to force the relay to be in the off position in it's default state, as specified by the "relayOnValue" on the config.json?

benlamonica, 2016-11-29:
Yes, that is a mistake in the docs. Will fix it shortly. Do you have a sensor to detect if the garage door is open or closed? Are you sure that sensor is working?

Did you already adjust your gpio pin in the OS to be active low?  

--Ben L

benlamonica, 2016-11-29:
I've uploaded 1.0.6 to add some logging to figure out what is going on. Will you install that and then paste your homebridge log into this thread?

fiddypal, 2016-11-29:
Yes i have an NC sensor which works just fine to display the status in the app.

What do you mean by "if i adjusted the gpio in the OS"? I have not made any changes to relay settings other than the toggle bits in the config.json file.

This is the type of relay board i am using: https://www.amazon.com/SunFounder-Channel-Shield-Arduino-Raspberry/dp/B00E0NTPP4/ref=as_li_ss_tl?s=pc&ie=UTF8&qid=1457587323&sr=1-3&keywords=raspberry+pi+relay&linkCode=sl1&tag=qui0d2-20&linkId=bee7b2fb14439e56441e942011d0cc55

There is a little red LED that lights up when the relay is triggered, that's how i can tell its "jammed on". 

config.json
```
        "accessories": [{
                "accessory": "RaspPiGPIOGarageDoor",
                "name": "Garage Door",
                "doorSwitchPin": 7,
                "relayOnValue": 0,
                "doorSwitchPressTimeInMs": 1000,
                "closedDoorSensorPin": 18,
                "closedDoorSensorValue": 0,
                "doorPollInMs": 4000,
                "doorOpensInSeconds": 14
        },
```

So when i reboot my system and homebridge starts up, the relay is triggered to ON (red led is lit) and the door opens. The relay stays ON (red led lit) until i press the button in the app. Then it cycles the relay for 1 second, and the relay from that point on stays in the OFF position at rest like it should.

It's like the default state upon boot is ON for the relay for some reason....

fiddypal, 2016-11-29:
updating to 1.0.6 now will paste logs in a sec

fiddypal, 2016-11-29:
for some reason when i do
```
npm update -g
```

It is not pulling down the latest files?

Thanks for bearing with me still a linux noob!

benlamonica, 2016-11-29:
Not sure why it's not updating, as it's published here: https://www.npmjs.com/package/homebridge-rasppi-gpio-garagedoor

For your problem though, It seems like you're having trouble with the gpio pin state on reboot, and this isn't really a homebridge issue.

http://raspberrypi.stackexchange.com/questions/32639/why-are-some-gpio-pins-high-when-the-raspberry-pi-boots-up

I believe there are some pins that default to HIGH when you start them (your relay board is set up to trigger on LOW I assume), so I would change the pins that you are using to use those to prevent unwanted triggering.

Another thread on this topic:

https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=24491

benlamonica, 2016-11-29:
by the way, 1.0.6 changes the config to be doorSwitchValue instead of relayOnValue. So make sure to update your config.json.

fiddypal, 2016-11-29:
I am not sure that is the issue Ben, the relay does not jam on when the pi boots but rather when homebridge fires up. 

I know this because i just switched it to GPIO 11 which starts as LOW, and as soon as i plugged in the PI the LED was lit on the relay board. If i leave it on GPIO 7 which starts as HIGH, the relay stays off when the pi is booted and it only gets jammed on when i launch homebridge.

Doing a fresh install now to try and pull down the latest 1.0.6 code base so i can share the logs with you.

fiddypal, 2016-11-29:
I think i have further narrowed this down... So this relay jamming only happens upon the first boot. If i press the open button to unjam it, shut down homebridge, then fire it back up again the relay does not jam on.

This leads me to believe the issue is happening maybe in the init.d script that's turning on the GPIO pins? How can i force a HIGH after the pins get set?

fiddypal, 2016-11-29:
@benlamonica Adding the following line after i set the relay pin to "OUT" in the init.d script seems to have solved this issue!

```
echo 0 > /sys/class/gpio/gpio7/value
```

is there a rouge echo somewhere setting it to 1 on first boot? Very strange...

benlamonica, 2016-11-29:
From https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=24491

> Just as a point of interest, the reason your three LEDs are on at start up is probably because five of the 17 available GPIO lines are pulled high by default (the rest are pulled low). The numbers are GPIO0/2, GPIO1/3, GPIO4, GPIO7 and GPIO8.

So if you pick a pin that is not one of those, it should not be an issue. However, sounds like you have solved the problem, so that's good.

closed 2016-11-29 by benlamonica