benlamonica/homebridge-rasppi-gpio-garagedoor #14: GPIO output goes to high on startup

alanfromtoronto, 2016-12-31:
When the garage-door-gpio script runs at startup it turns the pins to High which keeps the closed.  When I initiate an open or close the door it initiates the one second pulse but still keeps the circuit closed.  This is problematic due to the fact that if there is a power failure and the Pi restarts it automatically will open my garage and my multifunction control panel is not working as it is not expecting the line to be always on.

Help!

simonrb2000, 2016-12-31:
Mine doesn't behave that way. If my Pi loses power and turns back on or I physically reboot it, my garage door stays closed.

benlamonica, 2016-12-31:
It's due to the pins that you've chosen. They have a default state when the pi boots up. I'll find the docs and send it to you when I get to my computer. 

--Ben L

rosskPCS, 2016-12-31:
Can you add this here as well I am having the same issue. I am using pins 23 and 27 for the relay. When I start the script the relay fires and stays open until initiate a open or close command via homekit. Any help would be appreciated.

benlamonica, 2017-01-01:
I will add this information to the instructions. This is the reason why the pin is going high on reboot:

http://raspberrypi.stackexchange.com/questions/32639/why-are-some-gpio-pins-high-when-the-raspberry-pi-boots-up

I believe there are some pins that default to HIGH when you start them (your relay board is set up to trigger on LOW I assume), so I would change the pins that you are using to use those to prevent unwanted triggering.

Another thread on this topic:

https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=24491

benlamonica, 2017-01-01:
I've added additional information to the README on the GPIO pins that should be chosen, as well as some pictures on how to set up the sensor and relay. Let me know if picking different pins fixes your problem.

benlamonica, 2017-01-01:
See: https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor/blob/master/README.md

fiddypal, 2017-01-01:
I ran into this same issue and was able to get around it in software by modifying the startup init.d script to force the pin to low as soon as it's initialized.(or maybe it was high, I forget)

I believe another way to fix this on the hardware side is to swap GPIO pins like you suggested OR try swapping pins on your relay. Relay should have 3 pins, 1 is common and the other 2 are for normal open/ normal closed operation. So you can try connecting to the other pole on the relay and that should fix it as well.

fiddypal, 2017-01-01:
Found it:

Adding the following line after i set the relay pin to "OUT" in the init.d script seems to have solved this issue!

```
echo 0 > /sys/class/gpio/gpio7/value
```

fiddypal, 2017-01-01:
@benlamonica the default state of the GPIO pins that you reference, when does that action happen? I talked with some other users having this issue on slack and we determined the pins are getting set when your init.d script initializes them, NOT when the pi boots up. Are we taking about the same thing here?

benlamonica, 2017-01-01:
There appear to be 3 trouble times when the pins can flip. When the PI boots, and When the OS initializes the GPIO pins, and when the script exports the pins. We need to be aware of all three, otherwise it could trigger the garage door.

benlamonica, 2017-01-01:
@alanfromtoronto does this answer your question? Add a line like this to your gpio config script:
echo 0 > /sys/class/gpio/gpio7/value

where gpio7 should be changed to the gpio pin that you are using? Please re-open if you're having further troubles.

closed 2017-01-01 by benlamonica

alanfromtoronto, 2017-01-01:
This does answer my question and I changed the code to the following:
echo **in** > /sys/class/gpio/gpio23/direction
echo out > /sys/class/gpio/gpio23/direction
echo 0 > /sys/class/gpio/gpio23/value

As per the others I found that when I started up with the value as an input it does not pull up the value and then switching it to an output seemed to work

toshibochan, 2017-01-02:
Why 23 is IN and OUT? Not one or other?

rosskPCS, 2017-01-03:
The issue I am having is when I start the script either on startup or manually the first time the relay fires and stays on until I trigger the door via homebridge and it turns off the relay. From then on it work perfectly. I have tried adding echo 0 > /sys/class/gpio/gpio23/value but it has not helped. Tried adding in and out commands and it crashes hombridge the first time i trigger the door.

benlamonica, 2017-01-03:
@rosskPCS - will you paste the logs of the crash in a new issue and I'll take a look at it.

fiddypal, 2017-01-04:
Make sure you are declaring the IN and OUT pins correctly. Had the same issue with crashing because i was trying to write to an IN declared pin for the sensor instead of the OUT pin for the door relay.

Sent via mobile device