benlamonica/homebridge-rasppi-gpio-garagedoor #15: getting error when starting homebridge

toshibochan, 2016-12-31:
[Sat Dec 31 2016 14:26:04 GMT-0800 (PST)] ERROR LOADING PLUGIN homebridge-rasppi-gpio-garagedoor:
[Sat Dec 31 2016 14:26:04 GMT-0800 (PST)] Error: must run homebridge as root to control gpio pins
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:6:11)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Plugin.load (/usr/local/lib/node_modules/homebridge/lib/plugin.js:65:22)
    at Server.<anonymous> (/usr/local/lib/node_modules/homebridge/lib/server.js:128:14)
    at Array.forEach (native)
    at Server._loadPlugins (/usr/local/lib/node_modules/homebridge/lib/server.js:124:22)
    at new Server (/usr/local/lib/node_modules/homebridge/lib/server.js:48:24)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:26:16)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Function.Module.runMain (module.js:467:10)
    at startup (node.js:134:18)
    at node.js:961:3

benlamonica, 2017-01-01:
You need to run as root to control gpio pins. Please run as root.

benlamonica, 2017-01-01:
Was this working for you not as root before? I added that failure message because people were not running as root which is necessary in order to write to a GPIO pin...

toshibochan, 2017-01-01:
Before I never run on root! Strange

toshibochan, 2017-01-01:
Why now have open pin and close pin?

benlamonica, 2017-01-01:
It was a request made by someone else. You don't have to specify an open pin. It works fine with just a close pin. It also now works with no door sensor.

benlamonica, 2017-01-01:
I've fixed it in version 1.0.9 so that it only warns you when you're not running root, it doesn't fail to start.

closed 2017-01-01 by benlamonica

toshibochan, 2017-01-02:
Cool thanks!!!! Happy new year!!!!

commit fde4be38 (2017-01-01): fixes #15 - only warn when not root, don't fail to start.