benlamonica/homebridge-rasppi-gpio-garagedoor #16: open door sensor error

bozzomasterflash, 2017-01-07:
Hi, sorry for my bad english.
I set up 2 garage doors with these GPIO:
DOOR1: switch 23 - sensor 18
DOOR2: switch 22 - sensor 24
I have a problem: when I open the door, status switches from "closed" to opening and, 14 seconds later, to "open". Now, if I close Home app and reopen, status of the door is "No

 response" (see screenshot below) and the button doesn't work anymore...  If I ask "Siri close garage door", all works again...
What's wrong?

Here my config.json:
{
        "bridge": {
                "name": "piGarage",
                "username": "BD:52:5D:E3:CE:30",
                "port": 51826,
                "pin": "031-45-154"
        },

        "description": "This is an example configuration file with one fake acc$

        "accessories": [{
                "accessory": "RaspPiGPIOGarageDoor",
                "name": "Garage Stefano",
                "doorSwitchPin": 23,
                "doorSwitchPressTimeInMs": 500,
                "doorSwitchValue": 0,
                "closedDoorSensorPin": 18,
                "closedDoorSensorValue": 0,
                "openDoorSensorPin": 18,
                "openDoorSensorValue": 1,
                "doorPollInMs": 1000,
                "doorOpensInSeconds": 14
        }, {
                "accessory": "RaspPiGPIOGarageDoor",
                "name": "Garage Norino",
                "doorSwitchPin": 22,
                "doorSwitchPressTimeInMs": 500,
                "doorSwitchValue": 0,
                "closedDoorSensorPin": 24,
                "closedDoorSensorValue": 0,
                "openDoorSensorPin": 24,
                "openDoorSensorValue": 1,
                "doorPollInMs": 1000,
                "doorOpensInSeconds": 14
        }],

        "platforms": []
}

And here my garage-door-gpio:
#!/bin/sh
### BEGIN INIT INFO
# Provides:          garage-door-gpio
# Required-Start:    $remote_fs dbus
# Required-Stop:     $remote_fs dbus
# Should-Start:      $syslog
# Should-Stop:       $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Enable Garage Door GPIO
# Description:       Enable Garage Door GPIO
### END INIT INFO

PATH=/sbin:/bin:/usr/sbin:/usr/bin
DESC="Enable GPIO for the Garage Door Module"
NAME="garage-door-gpio"
SCRIPTNAME=/etc/init.d/$NAME

. /lib/lsb/init-functions

case "$1" in
    start)
        log_daemon_msg "Enabling Garage Door GPIO"
        success=0
        if [ ! -e /sys/class/gpio/gpio23 ] ; then
                echo 23 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio23/direction     
                echo 1 > /sys/class/gpio/gpio23/value
        fi

        if [ ! -e /sys/class/gpio/gpio18 ] ; then
                echo 18 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio18/direction
        fi

        if [ ! -e /sys/class/gpio/gpio24 ] ; then
                echo 24 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio24/direction
        fi

        if [ ! -e /sys/class/gpio/gpio22 ] ; then
                echo 22 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio22/direction
                echo 1 > /sys/class/gpio/gpio22/value
        fi
 
Can anyone help me?
Thank you
Stefano

bozzomasterflash, 2017-01-07:
[image]

daviderie, 2017-01-07:
I had a similar setup and realized that when you only have one pin to sense whether the door is open or closed, then you should only list that one in the config.json. I listed it as the openDoorSensorPin, but then found a few bugs in the plugin in that scenario. I fixed the bugs and just committed them to a branch and created a pull request for the author to merge them into the master. Here is what my config.json looks like:

`
    "accessories": [
	{
           "accessory": "RaspPiGPIOGarageDoor",
           "name": "Dave's Garage Door",
           "doorSwitchPin": 4,
           "doorSwitchPressTimeInMs": 1000,
           "doorSwitchValue": 0,
           "openDoorSensorPin": 27,
           "openDoorSensorValue": 0,
           "doorPollInMs": 4000,
           "doorOpensInSeconds": 14
       },
	{
           "accessory": "RaspPiGPIOGarageDoor",
           "name": "Karen's Garage Door",
           "doorSwitchPin": 17,
           "doorSwitchPressTimeInMs": 1000,
           "doorSwitchValue": 0,
           "openDoorSensorPin": 22,
           "openDoorSensorValue": 0,
           "doorPollInMs": 4000,
           "doorOpensInSeconds": 14
       }
    ]
`

daviderie, 2017-01-07:
Also, I found out that if your outputs to trigger the door opener should be normally high (as mine are due to the relay I'm using), then this is how your init.d script should look to accomplish that:

```
case "$1" in
    start)
        log_daemon_msg "Enabling Garage Door GPIO"
	success=0

	# Left garage door output (must be high to keep door closed)
	if [ ! -e /sys/class/gpio/gpio4 ] ; then
		echo 4 > /sys/class/gpio/export
		success=$?
		echo high > /sys/class/gpio/gpio4/direction
	fi

	# Right garage door output (must be high to keep door closed)
	if [ ! -e /sys/class/gpio/gpio17 ] ; then
		echo 17 > /sys/class/gpio/export
		success=$?
		echo high > /sys/class/gpio/gpio17/direction
	fi

	# Left garage door input
	if [ ! -e /sys/class/gpio/gpio27 ] ; then
		echo 27 > /sys/class/gpio/export
		success=$?
		echo in > /sys/class/gpio/gpio27/direction
	fi
 
	# Right garage door input
	if [ ! -e /sys/class/gpio/gpio22 ] ; then
		echo 22 > /sys/class/gpio/export
		success=$?
		echo in > /sys/class/gpio/gpio22/direction
	fi

        log_end_msg $success
        ;;
    *)
        echo "Usage: $SCRIPTNAME {start}" >&2
        exit 1
        ;;
esac
```

bozzomasterflash, 2017-01-07:
Hi @daviderie , thank you very much for your reply. I'll try to not list close sensor.
About the second suggestion, I'm wondering why I must set direction of the switch "high" instead of "out"... Which is the difference?
Thank's again for you help!
Stefano

daviderie, 2017-01-07:
@bozzomasterflash No problem...glad to help since I was having similar trouble. I found the "high" setting in this text file describing how the gpio stuff works: [https://www.kernel.org/doc/Documentation/gpio/sysfs.txt](url)

```
/sys/class/gpio/gpioN/

	"direction" ... reads as either "in" or "out". This value may
		normally be written. Writing as "out" defaults to
		initializing the value as low. To ensure glitch free
		operation, values "low" and "high" may be written to
		configure the GPIO as an output with that initial value.
```

benlamonica, 2017-01-08:
The open door sensor pin is there to detect when the door is fully open. The door closed sensor pin is there to detect when the door is fully closed. I only have a door close sensor pin (it's better to know that your door is closed than to care if it's fully open)

--Ben L

bozzomasterflash, 2017-01-08:
Hi @benlamonica thanks for your reply. 
So, if I have only The closed door sensor, should delete lines of opensensor in config.json, as suggested by @daviderie ?

benlamonica, 2017-01-08:
Correct. If you don't have the sensor, just remove the line.

--Ben L

bozzomasterflash, 2017-01-08:
OK, thank you all for the help!
Issue solved!!!
Ciao
Stefano

closed 2017-01-08 by bozzomasterflash