benlamonica/homebridge-rasppi-gpio-garagedoor #18: Having trouble with sensor

rex555, 2017-01-15:
I am having an issue with the plugin not reading my door state correctly. When it is closed it is fine, when I open it, it sets to open, then immediately sets to closed again. I have tried toggling the closedDoorSensorValue with no effect. Any thoughts?

When the door is open:
pi@raspberrypi ~ $ gpio read 12
1

When closed:
pi@raspberrypi ~ $ gpio read 12
0

My config:
{
    "bridge": {
        "name": "Homebridge",
        "username": "CC:22:3D:E3:CE:30",
        "port": 51826,
        "pin": "053-73-874"
    },

    "accessories": [{
        "accessory": "RaspPiGPIOGarageDoor",
        "doorSwitchPin": 7,
        "closedDoorSensorPin": 12,
        "closedDoorSensorValue": 0,
        "doorPollInMs": 4000,
        "doorOpensInSeconds": 18,
        "name": "Garage Door"
    }],

    "platforms": [
        {
            "platform" : "LifxLan",
            "name" : "LIFX"
        }
    ]
}

[Sun Jan 15 2017 00:23:09 GMT+0000 (UTC)] Homebridge is running on port 51826.
[Sun Jan 15 2017 00:23:10 GMT+0000 (UTC)] [LIFX] Online: Entrance 3 [d073d5208bfc]
[Sun Jan 15 2017 00:23:10 GMT+0000 (UTC)] [LIFX] Online: Entrance 1 [d073d5206f76]
[Sun Jan 15 2017 00:23:15 GMT+0000 (UTC)] [LIFX] Online: Entrance 2 [d073d5208bf3]
[Sun Jan 15 2017 00:23:32 GMT+0000 (UTC)] [Garage Door] GarageDoor is CLOSED (1)
[Sun Jan 15 2017 00:23:33 GMT+0000 (UTC)] [Garage Door] Setting state to 0
[Sun Jan 15 2017 00:23:33 GMT+0000 (UTC)] [Garage Door] Triggering GarageDoor Relay
[Sun Jan 15 2017 00:23:33 GMT+0000 (UTC)] [Garage Door] Turning on GarageDoor Relay, pin 7 = 1
[Sun Jan 15 2017 00:23:34 GMT+0000 (UTC)] [Garage Door] Turning off GarageDoor Relay, pin 7 = 0

** Right here **
[Sun Jan 15 2017 00:23:51 GMT+0000 (UTC)] [Garage Door] Set current state to OPEN
[Sun Jan 15 2017 00:23:53 GMT+0000 (UTC)] [Garage Door] Door state changed to CLOSED
[Sun Jan 15 2017 00:24:19 GMT+0000 (UTC)] [Garage Door] Setting state to 1
[Sun Jan 15 2017 00:24:34 GMT+0000 (UTC)] [Garage Door] GarageDoor is CLOSED (1)
[Sun Jan 15 2017 00:24:35 GMT+0000 (UTC)] [Garage Door] Setting state to 0
[Sun Jan 15 2017 00:24:35 GMT+0000 (UTC)] [Garage Door] Triggering GarageDoor Relay
[Sun Jan 15 2017 00:24:35 GMT+0000 (UTC)] [Garage Door] Turning on GarageDoor Relay, pin 7 = 1
[Sun Jan 15 2017 00:24:36 GMT+0000 (UTC)] [Garage Door] Turning off GarageDoor Relay, pin 7 = 0

** Right here **
[Sun Jan 15 2017 00:24:53 GMT+0000 (UTC)] [Garage Door] Set current state to OPEN
[Sun Jan 15 2017 00:24:57 GMT+0000 (UTC)] [Garage Door] Door state changed to CLOSED
[Sun Jan 15 2017 00:25:07 GMT+0000 (UTC)] [Garage Door] Setting state to 1
[Sun Jan 15 2017 00:26:25 GMT+0000 (UTC)] [Garage Door] GarageDoor is CLOSED (1)
[Sun Jan 15 2017 00:31:26 GMT+0000 (UTC)] [Garage Door] GarageDoor is CLOSED (1)
[Sun Jan 15 2017 00:31:28 GMT+0000 (UTC)] [Garage Door] Setting state to 0
[Sun Jan 15 2017 00:31:28 GMT+0000 (UTC)] [Garage Door] Triggering GarageDoor Relay
[Sun Jan 15 2017 00:31:28 GMT+0000 (UTC)] [Garage Door] Turning on GarageDoor Relay, pin 7 = 1
[Sun Jan 15 2017 00:31:29 GMT+0000 (UTC)] [Garage Door] Turning off GarageDoor Relay, pin 7 = 0

** Right here **
[Sun Jan 15 2017 00:31:46 GMT+0000 (UTC)] [Garage Door] Set current state to OPEN
[Sun Jan 15 2017 00:31:46 GMT+0000 (UTC)] [Garage Door] Door state changed to CLOSED
[Sun Jan 15 2017 00:32:51 GMT+0000 (UTC)] [Garage Door] GarageDoor is CLOSED (1)

toshibochan, 2017-01-15:
Try change "closedDoorSensorValue": 1

rex555, 2017-01-15:
Same behavior. It does not seem to be reading the sensor properly

toshibochan, 2017-01-16:
You did this part?
Export the GPIO pins to be used and set their direction after reboot
a. Copy and edit this start script into your /etc/init.d directory.
b. Change the values to be the gpio pins that you are using.
c. chmod 755 /etc/init.d/garage-door-gpio # this makes the script executable
d. sudo update-rc.d /etc/init.d/garage-door-gpio defaults # this will set up the symlinks to run the script on startup.
e. sudo /etc/init.d/garage-door-gpio start and verify that your pins are exported by looking in ls /sys/class/gpio/ directory

toshibochan, 2017-01-16:
This is my  switch pin:

       if [ ! -e /sys/class/gpio/gpio19 ] ; then
                echo 19 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio19/direction
                echo 1 > /sys/class/gpio/gpio19/active_low
                echo 0 > /sys/class/gpio/gpio19/value
        fi

And this is my sensor:

      if [ ! -e /sys/class/gpio/gpio21 ] ; then
                echo 21 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio21/direction
        fi

rex555, 2017-01-16:
Yeah I export the pins using 'gpio export <pin num> out' for the garage door controller pin, and 'gpio export <pin num> in' for the sensor pin.
Which sensor pin num do I usr though? The door pin I was lucky enough to use 7 which is 7 in wPi, 7 Physical and GPIO.7. Which number do I use for the config file for the sensor? Physical, Name, or wPi?

I have not even mucked with the startup scripts yet but the commands do the same thing. Switch pin works fine, it is the sensor pin not working properly or at odds with reality

rex555, 2017-01-16:
Ok, I think I got it, this is using the BCM pinout table. I had the wrong pin for sensors. Thanks for the help!

toshibochan, 2017-01-16:
> Yeah I export the pins using 'gpio export out' for the garage door controller pin, and 'gpio export in' for the sensor pin.

My is GPIO19 pin 35 and GPIO21 pin 40

toshibochan, 2017-01-16:
You welcome!!!

benlamonica, 2017-01-16:
@rex555 glad you got it working. Feel free to re-open this issue if you have further problems.

@toshibochan thank you so much for helping people who are having problems. I really appreciate your advice to those trying to get it all working 👍

closed 2017-01-16 by benlamonica

toshibochan, 2017-01-16:
@benlamonica thank you!!! If not you guys for make this greats homebridge plugins for us we still just pressing the switches manually!! 
Thanks and thanks

toshibochan, 2017-01-16:
Now I waiting for some one make fan plugin using relay to connect to may fan remote control

toshibochan, 2017-01-16:
This is  example for the  ceiling fan  remote control and relay. Work with Siri but she don't use homebridge.
https://youtu.be/bFAyaKc0ChU

benlamonica, 2017-01-16:
I've actually been wanting to automate my fans..will have to take a look at that.

toshibochan, 2017-01-16:
Let me know when you finish the plugin!!!!

rex555, 2017-01-17:
Hey guys, one more question. When I export the pins with gpio the wired garage door button ceases to work, any idea if this is supposed to happen? The hap-nodejs direct implementation I was using before used python to export the pin, trigger it, and unexport the pin each time you asked for an open or close so it was not an issue. Any way to do that here?

toshibochan, 2017-01-17:
Show me your file

fiddypal, 2017-01-17:
Initialization of GPIO pins happens in the init.d script. Make sure you are exporting the pins correctly, if you try to export the relay as 'IN' then it will act like a sensor and not change when you press the button.

Sent via mobile device

rex555, 2017-01-18:
#!/bin/sh
### BEGIN INIT INFO
# Provides:          garage-door-gpio
# Required-Start:    $remote_fs dbus
# Required-Stop:     $remote_fs dbus
# Should-Start:	     $syslog
# Should-Stop:       $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Enable Garage Door GPIO
# Description:       Enable Garage Door GPIO
### END INIT INFO

PATH=/sbin:/bin:/usr/sbin:/usr/bin
DESC="Enable GPIO for the Garage Door Module"
NAME="garage-door-gpio"
SCRIPTNAME=/etc/init.d/$NAME

. /lib/lsb/init-functions

case "$1" in
    start)
        log_daemon_msg "Enabling Garage Door GPIO"
        success=0
        if [ ! -e /sys/class/gpio/gpio7 ] ; then
            echo 7 > /sys/class/gpio/export
            success=$?
            echo out > /sys/class/gpio/gpio7/direction
        fi

        if [ ! -e /sys/class/gpio/gpio18 ] ; then
            echo 18 > /sys/class/gpio/export
            success=$?
            echo in > /sys/class/gpio/gpio18/direction
        fi
 
        log_end_msg $success
        ;;
    *)
        echo "Usage: $SCRIPTNAME {start}" >&2
        exit 1
        ;;
esac

exit 0

rex555, 2017-01-18:
eh, that ended up ugly

fiddypal, 2017-01-18:
7 should be your relay and 18 your sensor.

Sent via mobile device

fiddypal, 2017-01-18:
Echoing 1/0 just turns the relay on or off, however that will not work correctly if you are trying to echo out to a pin that's been configured to read and not write (in/out).