benlamonica/homebridge-rasppi-gpio-garagedoor #19: Close notification  issue

toshibochan, 2017-01-19:
When I ask Siri to close the garage I receive this: "garage was opened" and after close say "garage was closed" when I open show correct but just when I close the first notification show opened inter closed.
Any one having this problem?

benlamonica, 2017-01-19:
Have you upgraded to the latest iOS? That problem went away after I upgraded. 

--Ben L

toshibochan, 2017-01-19:
Yes. My iOS is in 10.2

toshibochan, 2017-01-19:
All devices we have do the same

benlamonica, 2017-01-22:
What is your config? Will you paste in your config and also a snippet of the log with it receiving an open command?

benlamonica, 2017-01-22:
Also, what version of the plugin and home bridge are you using?

toshibochan, 2017-01-24:
I not home till march. I will sent you wen I back home. Thanks for helping