benlamonica/homebridge-rasppi-gpio-garagedoor #22: Adding a physical push button switch

KevP, 2017-02-11:
Is it possible to add a momentry push button/switch that will also open/close the garage door? Like the old push button that would have been in place but it triggers through Homebridge to open/close the garage door? Good work BTW!

benlamonica, 2017-02-11:
If you look at the pictures in the README, I didn't actually remove the button. I just hooked the wires for the relay in at the same point as the button normally connects. So my button is still on my wall and can open and close the garage door, and homebridge just sends me notifications when the garage opens/closes. So you should be able to keep the button without issue.

KevP, 2017-02-12:
Ok great, thank you.

closed 2017-02-12 by KevP