benlamonica/homebridge-rasppi-gpio-garagedoor #23: Help with GPIO Pin Permissions

simonrb2000, 2017-02-12:
Hi all, I think I know what the problem is but can't seem to find a way to fix it.

I don't think I have permissions and I don't think I run Homebridge as root. When I try to the accessories don't show up. So when I try to open my garage door I get this error

```
pi@raspberrypi ~ $ tail -f /var/log/homebridge.err
    at Bridge.Accessory._handleSetCharacteristics (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:672:8)
    at emitMany (events.js:108:13)
    at HAPServer.emit (events.js:182:7)
    at HAPServer._handleCharacteristics (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:967:10)
    at HAPServer.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:204:39)
    at emitNone (events.js:67:13)
    at IncomingMessage.emit (events.js:166:7)
    at endReadableNT (_stream_readable.js:893:12)
    at doNTCallback2 (node.js:429:9)
    at process._tickCallback (node.js:343:17)
```

If I run these commands at terminal gpio export 14 in & gpio export 4 out BEFORE I try to open the garage then all works as expected. Unfortunately sometimes my Pi restarts then when someone tries to open the garage door the above error happens. I have made a script with the above commands in and if I run that from terminal then it all works. I just can't find a way to run said script automatically at boot up. I have tried various ways, including the crontab. Any ideas?

Cheers!!!

benlamonica, 2017-02-13:
How are you starting homebridge when the raspberry pi restarts? Add the commands to that script. 

--Ben L

simonrb2000, 2017-03-05:
I am using systemd to start it with a homebridge.service script. I can add it to that? It's the default way to do it as shown on the wiki. 

Cheers!

Kulen01, 2018-05-09:
You need to add the homebridge user you created for the systemd startup to the gpio group.

usermod -a -G gpio homebridge to add the homebridge user to the gpio group.