benlamonica/homebridge-rasppi-gpio-garagedoor #25: DoorOpenSensor

simonrb2000, 2017-03-10:
Hi all,

I got the idea of adding a reed switch to detect the door when it is open from your readme. I have installed it and it works fine on the raspberry pi and gpio user has access to the pin. 1 is when its open and 0 is when its closed. 

I have added the required line to the config and homebridge starts fine. The issue I am getting is when the garage door is opened from the wall switch or from any means other than homekit then homekit comes up with no response when its open. So I can't close it. Normal situation is pulling up in the car and opening the garage using the opener in the car, then once car is in the garage I use Siri to close it (don't have to carry an opener with me that way). 

Any ideas? Maybe my pin is the wrong way around, high and low. I will try that. Any other ideas?

benlamonica, 2017-03-10:
Would you post your logs of when the door opens? It's very possible that there is a bug in that code because I don't have an open door sensor (I only have a closed door sensor) so I haven't really tested it.

simonrb2000, 2017-03-10:
I probs. How do I get the logs? Sorry never had to debug before

simonrb2000, 2017-03-10:
Can't find logs anywhere. I am running via systemctl. This is what I have found. Shows as door STOPPED - thats when homekit appears to lose connection

Mar 10 22:35:26 raspberrypi homebridge[18420]: [3/10/2017, 10:35:26 PM] [Garage Door] GarageDoor is OPEN (0)
Mar 10 22:35:27 raspberrypi homebridge[18420]: [3/10/2017, 10:35:27 PM] [Garage Door] Setting state to 1
Mar 10 22:35:27 raspberrypi homebridge[18420]: [3/10/2017, 10:35:27 PM] [Garage Door] Triggering GarageDoor Relay
Mar 10 22:35:27 raspberrypi homebridge[18420]: [3/10/2017, 10:35:27 PM] [Garage Door] Turning on GarageDoor Relay, pin 4 = 1
Mar 10 22:35:28 raspberrypi homebridge[18420]: [3/10/2017, 10:35:28 PM] [Garage Door] Turning off GarageDoor Relay, pin 4 = 0
Mar 10 22:35:47 raspberrypi homebridge[18420]: [3/10/2017, 10:35:47 PM] [Garage Door] Set current state to CLOSED
Mar 10 22:36:21 raspberrypi homebridge[18420]: [3/10/2017, 10:36:21 PM] [Garage Door] Door state changed to STOPPED
Mar 10 22:36:36 raspberrypi homebridge[18420]: [3/10/2017, 10:36:36 PM] [Garage Door] GarageDoor is OPEN (0)
Mar 10 22:37:07 raspberrypi homebridge[18420]: [3/10/2017, 10:37:07 PM] [Garage Door] GarageDoor is STOPPED (4)
Mar 10 22:37:17 raspberrypi homebridge[18420]: [3/10/2017, 10:37:17 PM] [Garage Door] Door state changed to CLOSED

benlamonica, 2017-03-11:
Thanks, what you pasted was the logs, that's what I wanted. I'll take a look at the code and see what I can find.

simonrb2000, 2017-03-11:
Thanks, anything you require let me know.

simonrb2000, 2017-04-28:
Any ideas on this?

tunip, 2018-06-15:
Hi,

i have the same issue as simonrb2000 with an additional opener sensor.

Would be fine to use HomeKit AND a wall swtich together with correct states/function.