benlamonica/homebridge-rasppi-gpio-garagedoor #28: Enhancement:  Gate Control

ansuz07, 2017-05-30:
First off- thanks for the great plugin.  I have it working on my garage and it is flawless.

My question for you is how hard would it be to have a version that didn't look for open/close sensors and just automatically defaulted to "closed" after a few seconds?

The reason I ask is because I have both a perimeter access gate and a garage door.  The gate is too far away to wire up an open/closed sensor and it really doesn't matter since the gate automatically closes after about 30 seconds.  It would be great if I could use the plugin to open the gate and have it just default back to closed after a set amount of time.

benlamonica, 2017-05-31:
It should already do that, though I just merged a bug fix that was submitted by someone and haven't published it to npm yet. 

But in your configure, try just leaving the sensor pins out and it should still work as you want. 

-Ben L

benlamonica, 2017-06-11:
Oh, I see what you are asking about. You want it to automatically fall back to closed after a specific timeout. Yes, I think we can add that. I'm sorry that I've been so slow lately at responding.

If you're up to the challenge, you could make the change yourself, by adding a new config to the index.js and if it's set, then set a timer after a button is pressed to change the state back to the original state.

Or I can do it when I get to it :)

toshibochan, 2017-06-11:
How far is your gate?

ansuz07, 2017-06-12:
@benlamonica - Exactly.  Sometimes you don't need the open monitor (like with an automatic gate) but would still like to be able to automate it to open (like when you cross a geo-fence boundary).  No worries on the late response; we all have lives.  I would take a look at the file, but my coding skills max out at a config.json configuration, sadly.

@toshibochan - far enough away to make running a physical wire problematic.  Right now, I have the opener wired up to a delay switch that toggles current between the opener and the open monitor so "simulate" an open state for about 10 seconds before closing.  Not a perfect fix, but functional enough.

toshibochan, 2017-06-12:
Wifi reach in your gate? You can somehow put raspberry pi zero in your gate?

ansuz07, 2017-06-12:
Opening the gate isn't the issue - I have a remote wired up that can handle that. Monitoring the state is the issue. 

I could put another box dedicated to just the gate, but that is more investment than I want to make, especially since I don't actually care about the open/closed state

tigroz, 2017-06-27:
> If you're up to the challenge, you could make the change yourself, by adding a new config to the index.js and if it's set, then set a timer after a button is pressed to change the state back to the original state.

I'm interested in this too! please, can you explain ? 
how to have a version that didn't look for open/close sensors and just automatically defaulted to "closed" after a few seconds? I also  have the opener wired up to a delay switch ...

tigroz, 2017-07-03:
@ansuz07 what kind of delay switch have you used? have a foto?

ansuz07, 2017-07-06:
This is the one I ended up using:

https://www.amazon.com/gp/product/B01LVTOW0L/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

tigroz, 2017-07-06:
@ansuz07 tnx, how did you have connected? anyway, it could be better implement a timer in the index.js ( but i don't know how!)

ansuz07, 2017-07-06:
I have it wired up where the Pi triggers the delay switch directly.  The resting state supplies power to the door close sensor pin and the triggered state is wired directly to the gate opener with a 10s delay