benlamonica/homebridge-rasppi-gpio-garagedoor #29: Status stays Opening/Closing at Home.app in IOC 10.3.2

xiaoqiangwang, 2017-06-10:
I am new to homebridge platform and intend to use it to control the garage door.I  basically used the same config as in the example. PIN 18 for trigger, PIN 24 and 25 for Close/Open input. 

I simulate the close/open sensor by connecting them to 3.3V. On the console log, it reports correctly the state changes.
[2017-6-10 18:46:31] [Garage Door] GarageDoor is OPEN (0)
[2017-6-10 18:46:38] [Garage Door] Setting state to CLOSE
[2017-6-10 18:46:38] [Garage Door] Triggering GarageDoor Relay
[2017-6-10 18:46:38] [Garage Door] Turning on GarageDoor Relay, pin 18 = 1
[2017-6-10 18:46:40] [Garage Door] Turning off GarageDoor Relay, pin 18 = 0
[2017-6-10 18:46:53] [Garage Door] Set current state to CLOSED

But the Home app stuck in showing  "Closing". Any observation of such issue?

benlamonica, 2017-06-11:
do you have 2 sensors set up? If you don't have an open sensor, you should just not include that in the config.

xiaoqiangwang, 2017-06-11:
I have simulated the two sensors scenario by short-circuit the Closed(Pin 24) / Open(Pin25). As you can see from the log, the the plugin recognise the state has reached CLOSED.

Only that the Home app does not poll the new status.

simonrb2000, 2017-06-11:
I couldn't get this to work properly with two sensors (had one for fully open and one fully closed). I ended up disabling the fully open one just to get it to work.

xiaoqiangwang, 2017-06-11:
@simonrb2000 Thanks for the info. I made a test with only CLOSE sensor
```
{
    "bridge": {
        "name": "Homebridge",
        "username": "CC:22:3D:E3:CE:30",
        "port": 51828,
        "pin": "031-45-154"
    },
    
    "description": "RaspberryPi HomeBridge",

     "accessories": [{
           "accessory": "RaspPiGPIOGarageDoor",
           "name": "Garage Door",

           "doorSwitchPin": 18,
           "doorSwitchPressTimeInMs": 2000,
           "doorSwitchValue": 1,

           "closedDoorSensorPin": 24,
           "closedDoorSensorValue": 1,

           "doorPollInMs": 4000,
           "doorOpensInSeconds": 15
       }]
}
```

Here is the plugin output
```
Jun 11 18:10:32 wangsrpi homebridge[832]: [2017-6-11 18:10:32] [Garage Door] Triggering GarageDoor Relay
Jun 11 18:10:32 wangsrpi homebridge[832]: [2017-6-11 18:10:32] [Garage Door] Turning on GarageDoor Relay, pin 18 = 1
Jun 11 18:10:34 wangsrpi homebridge[832]: [2017-6-11 18:10:34] [Garage Door] Turning off GarageDoor Relay, pin 18 = 0
Jun 11 18:10:41 wangsrpi homebridge[832]: [2017-6-11 18:10:41] [Garage Door] Setting state to CLOSE
Jun 11 18:10:47 wangsrpi homebridge[832]: [2017-6-11 18:10:47] [Garage Door] Set current state to CLOSED
```

PIN 24 is indeed already active,  but the Home app shows closing.
```
$ cat /sys/class/gpio/gpio24/value 
1
```
[image]

toshibochan, 2017-06-11:
Did you make garage-door-gpio in the /etc/init.d/?

xiaoqiangwang, 2017-06-11:
I modified the example /etc/init.d/garage-door-gpio for my setup. 
```
#!/bin/sh
### BEGIN INIT INFO
# Provides:          garage-door-gpio
# Required-Start:    $remote_fs dbus
# Required-Stop:     $remote_fs dbus
# Should-Start:	     $syslog
# Should-Stop:       $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Enable Garage Door GPIO
# Description:       Enable Garage Door GPIO
### END INIT INFO

PATH=/sbin:/bin:/usr/sbin:/usr/bin
DESC="Enable GPIO for the Garage Door Module"
NAME="garage-door-gpio"
SCRIPTNAME=/etc/init.d/$NAME

. /lib/lsb/init-functions

case "$1" in
    start)
        log_daemon_msg "Enabling Garage Door GPIO"
	success=0
	if [ ! -e /sys/class/gpio/gpio18 ] ; then
		echo 18 > /sys/class/gpio/export
		success=$?
                echo 1 > /sys/class/gpio/gpio18/active_low
		echo out > /sys/class/gpio/gpio18/direction
                echo 0 > /sys/class/gpio/gpio18/value
	fi

	if [ ! -e /sys/class/gpio/gpio24 ] ; then
		echo 24 > /sys/class/gpio/export
		success=$?
		echo in > /sys/class/gpio/gpio24/direction
	fi
	
	if [ ! -e /sys/class/gpio/gpio25 ] ; then
		echo 25 > /sys/class/gpio/export
		success=$?
		echo in > /sys/class/gpio/gpio25/direction
	fi
 
        log_end_msg $success
        ;;
    *)
        echo "Usage: $SCRIPTNAME {start}" >&2
        exit 1
        ;;
esac

exit 0
```

Here is the gpio status as reported by ``gpio``
```
$ gpio readall
 +-----+-----+---------+------+---+---Pi 2---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 1 | OUT  | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 1 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 2---+---+------+---------+-----+-----+
```

toshibochan, 2017-06-11:
In my case 20 is the switch and 13 is the sensor

       if [ ! -e /sys/class/gpio/gpio20 ] ; then
                echo 20 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio20/direction
                echo 1 > /sys/class/gpio/gpio20/active_low
                echo 0 > /sys/class/gpio/gpio20/value

        fi

        if [ ! -e /sys/class/gpio/gpio13 ] ; then
                echo 13 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio13/direction

toshibochan, 2017-06-11:
Try do
sudo update-rc.d garage-door-gpio defaults

toshibochan, 2017-06-11:
Change
"doorPollInMs": 4000,
To 
"doorPollInMs": 7000,

toshibochan, 2017-06-11:
Work?

xiaoqiangwang, 2017-06-11:
No progress. I am reinstalling the nodejs/homebridge/homebridge-rasppi-gpio-garagedoor. And got "Segmentation fault" on start.

My nodejs is node-v8.1.0-linux-armv7l downloaded from web and installed home bridge from github.
```
npm install git+https://github.com/nfarina/homebridge.git
npm  install git+https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor.git
```

What are the versions on your working setup?

toshibochan, 2017-06-11:
You can install doing
sudo npm install -g homebridge
sudo npm install -g homebridge-rasppi-gpio-garagedoor

xiaoqiangwang, 2017-06-11:
That was my old setup. Which nodejs version?

toshibochan, 2017-06-11:
I'm using raspberry pi zero
Node v4.2.1

xiaoqiangwang, 2017-06-11:
Now it works properly with the original two sensors configuration. I only have to downgrade nodejs from v8.1.0 to v4.3.2 (which is the minimum version required by homebridge 0.4.20). 

Thanks all for the help!

closed 2017-06-11 by xiaoqiangwang

toshibochan, 2017-06-11:
Glad working now!

benlamonica, 2017-06-11:
Need to find out what's up with the newer versions of node. 

-Ben L