benlamonica/homebridge-rasppi-gpio-garagedoor #30: Triggering Relay but Garage Door Doesn't Open

jpaquino3, 2017-07-16:
First of all thank you for creating this plugin! I've followed your readme and have gotten to the point where when I press the garage door button in HomeKit the relay fires but my garage door doesn't open. I have my relay module wired for NO and the light goes on for 1 second. Am I missing something in the wiring process? I've seen so many different wiring diagrams on how to do this. Any help would be much appreciated. I've also included pictures and config for reference. 

[https://imgur.com/a/SV2aM](https://imgur.com/a/SV2aM)

    "accessories": [
       {
           "accessory": "RaspPiGPIOGarageDoor",
           "name": "Garage",
           "doorSwitchPin": 23,
           "doorSwitchPressTimeInMs": 1000,
           "doorSwitchValue": 1,
           "closedDoorSensorPin": 24,
           "closedDoorSensorValue": 1,
           "openDoorSensorPin": 25,
           "openDoorSensorValue": 1,
           "doorPollInMs": 4000,
           "doorOpensInSeconds": 14
       }

benlamonica, 2017-07-16:
So you hear the relay trigger, but the door doesn’t open? When you take the two wires going into the relay and just touch them together and then take them apart does your garage open?

-Ben L

jpaquino3, 2017-07-16:
Yes, if i take the two wires from the garage door (picture 3 red and yellow) to the relay (picture 2)  and touch them together, the door opens. When use homekit i press the garage door button and the relay triggers for one second (light on the relay illuminates, then turns off). The physical button to the garage still works just not the virtual button through homekit.

benlamonica, 2017-07-16:
I’m suspicious of the relay. Do you have a multimeter that you could probe the relay? If you have a continuity testing mode (when it beeps if the two probes touch) and then put each probe on the two contacts on the relay and then trigger the button through homekit. 

Can you take a picture of the relay board and the wiring?

-Ben L

benlamonica, 2017-07-16:
Oh, sorry. I see the pictures now. That all looks right. I’d try using the other relay, and then if that doesn’t work, try playing with the amount of time that the relay is kept open by increasing the doorSwitchPressTimeInMs.

jpaquino3, 2017-07-16:
It's has 2 relays and I tried both sides without luck I also increased the time to 8 seconds (8000) and it didn't work. Right now I don't have a multimeter but I'll go ahead and get one grab one so I can test everything out. 

Do you think that there isn't enough power going to the garage door opener for it to trigger? 

Thank you for the help!

benlamonica, 2017-07-16:
Let us know how it goes. 

The power is coming from the garage Door Opener itself. All the relay is doing is connecting the two wires together. The garage door opener is just supposed to detect when the wires touch.

If you hear it click, then the relay is working. But maybe it’s defective somehow. Only way to tell is find something that you can tell when the circuit is complete when the relay is on. 

-Ben L

jpaquino3, 2017-07-17:
The relay has never made a noise, just turns on the light. Also, now the light is stays illuminated until I activate the HomeKit button - which is opposite from the last time I booted. So most likely a defective relay considering there isn't an audible click?

toshibochan, 2017-07-17:
Did you setup garage-door-gpio file in /etc/unit.d/?

toshibochan, 2017-07-17:
Sorry /etc/init.d

jpaquino3, 2017-07-17:
@toshibochan Yes I did. The only thing I didn't do was set everything to run on start, since it wasn't working correctly, but I did start everything manually as root. I ordered another relay module and a multimeter. Everything is leaning towards a bad relay.

benlamonica, 2017-07-17:
There should definitely be an audible click when the relay energizes, which you should be able to hear if the room is quiet enough.

http://www.wikihow.com/Test-a-Relay

jpaquino3, 2017-07-19:
@benlamonica As you suspected it was a bad relay. Everything is fixed and working now. Thanks for the help and let me know if you'd like me to do anything to pitch in!

closed 2017-07-19 by jpaquino3

benlamonica, 2017-07-19:
Great! I’m glad you got it working.

-Ben L