benlamonica/homebridge-rasppi-gpio-garagedoor #31: Plugin and Eve Door/Window Sensor communication

GatoPharaoh, 2017-07-27:
Hi

I'm using an Elgato Eve Door and Window Sensor (which is HomeKit capable) to determine if my garage is open or closed.
Any idea how to implement the state of the Eve Sensor into the plugin. So that the plugin switch knows from the Eve sensor when my garage is opened or closed?

Thanks in advance

benlamonica, 2017-07-27:
Just brainstorming here, but maybe you could write a program to run on your raspberry pi that flips the state of a GPIO when the garage door is open/closed and then you can use the garagedoor plugin as is. Or you can modify the garagedoor plugin to query a different file which contains the state. Or you could just buy the $4 reed switch and hook it up and have 2 different door open sensors.

GatoPharaoh, 2017-07-27:
awesome input - thank you!

closed 2017-07-27 by GatoPharaoh