benlamonica/homebridge-rasppi-gpio-garagedoor #33: Automatically close garage after 10 mins

alanfromtoronto, 2017-08-27:
I would like to be able to automatically trigger a close after 10 mins, assuming it is open.  any idea how to do this?

benlamonica, 2017-08-27:
I have people who have requested that. It’s a bit dangerous though. When the door is opened, we can just start a timer that is configured in the config value. 

-Ben L

alanfromtoronto, 2017-08-27:
That works for me.  I am fine assuming that the automatic door sensor works and won't hurt anyone.

alanfromtoronto, 2017-09-06:
Thoughts on where we could include the timer?  I wouldn't mind being a beta tester :)

benlamonica, 2017-09-16:
I'm sorry I've been so unresponsive. Super busy. I hope to do some work soon, so I'll add this feature and update this ticket.

toshibochan, 2017-09-16:
Did you make garage-door-gpio file in /etc/init.d/?

simonrb2000, 2017-09-16:
Excellent.

Have you managed to test having two sensors? I have one when the door is fully open and fully closed. I can only use one or the other. If both are enabled it crashes.

Many thanks.

alanfromtoronto, 2017-09-17:
Ben - you may want to hold off on the timer as that should be covered off by the timers automation feature within iOS11 assuming you have an AppleTV.

tigroz, 2017-10-13:
Iupgraded to IOS 11 but still can't find timers automation,  Ben, can you please add a timer?

benlamonica, 2017-10-13:
Yes, will get to it soon. 

-Ben L

alanfromtoronto, 2018-10-06:
used a dummy switch and leveraged timers.  not totally smooth but all works.