benlamonica/homebridge-rasppi-gpio-garagedoor #34: Switching from old HAP-NodeJS config... Easy to do?

brettstah, 2017-09-15:
So for my very first raspberry pi project, I followed a good YouTube video to allow me to control the garage door via Homekit.  Works great, but... I really want to switch to a Homebridge based solution. 

Can I use the current wiring with no issues?

Relay module (wired to the same wires that connect to the garage door opener on the garage wall):
pin 26, gpio 7

Magnetic reed switch (to determine if door is closed/not closed):
pin 12, gpio 18

bbarbierLW, 2017-09-15:
OK, so I had some time today to try it - I am THIS close to having it work perfectly. Any thoughts/advice on what needs to be adjusted?

With the latest Raspian/nodejs/homebridge installed, and this plugin installed, I followed the installation steps. I changed the gpio pins 23 and 24 to match what my raspberry pi is actually using (7 for the relay, 18 for the reed switch).  

From the Home app, I can successfuly open and close the garage door, and the status of the door is accurately updated.

Here are the issues:

1) When the raspberry pi reboots, the garage door opens. Something to do with "default high" with the pin?  Could I simply change from using gpio 7 to 23 for the relay switch to resolve this? Or could I modify the way the raspberry pi boots up and deals with gpio 7 to leave the door closed (or open)?

2) Could be related to the above - but as soon as the export pin script executes, the red light on the button that normally controls the garage door the "old-school" way (it was installed by the builder, and is on the wall) goes out, and it ceases to function (it won't activate the garage door). At that point, I can ONLY control the garage door via HomeKit.  Any thoughts on that?

Thanks for the plugin! It's very slick and I hope to get these two kinks worked out...

brettstah, 2017-09-15:
OK, so I had some time today to try it - I am THIS close to having it work perfectly. Any thoughts/advice on what needs to be adjusted?

With the latest Raspian/nodejs/homebridge installed, and this plugin installed, I followed the installation steps. I changed the gpio pins 23 and 24 to match what my raspberry pi is actually using (7 for the relay, 18 for the reed switch).  

From the Home app, I can successfuly open and close the garage door, and the status of the door is accurately updated.

Here are the issues:

1) When the raspberry pi reboots, the garage door opens. Something to do with "default high" with the pin?  Could I simply change from using gpio 7 to 23 for the relay switch to resolve this? Or could I modify the way the raspberry pi boots up and deals with gpio 7 to leave the door closed (or open)?

2) Could be related to the above - but as soon as the export pin script executes, the red light on the button that normally controls the garage door the "old-school" way (it was installed by the builder, and is on the wall) goes out, and it ceases to function (it won't activate the garage door). At that point, I can ONLY control the garage door via HomeKit.  Any thoughts on that?

Thanks for the plugin! It's very slick and I hope to get these two kinks worked out...

brettstah, 2017-09-16:
OK, had some time just now to change the wiring of the relay module from gpio 7 to gpio 23. I updated the files where I had previously replaced 23 with 7, rebooted, and the garage door opened as soon as the init.d script ran. :\   Also, the wall button light went out, and it was unable to activate the garage door. As soon as I power down the raspi, the light comes on and the button works again.

Hopefully I'm just missing something simple! Any help would be appreciated!

toshibochan, 2017-09-16:
Did you make garage-door-gpio file in /etc/init.d/?

toshibochan, 2017-09-16:
in my case my button is 21 and relay is 20.
change 21 to you button number and 20 to your relay number.

sudo nano /etc/init.d/garage-door-gpio
*copy and past my script and change the 21 and 20 to your number and save.
sudo chmod 755 /etc/init.d/garage-door-gpio
sudo update-rc.d garage-door-gpio defaults
sudo /etc/init.d/garage-door-gpio start

now try start homebridge

my garage-door-gpio:

#!/bin/sh
### BEGIN INIT INFO
# Provides:          garage-door-gpio
# Required-Start:    $remote_fs dbus
# Required-Stop:     $remote_fs dbus
# Should-Start:      $syslog
# Should-Stop:       $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Enable Garage Door GPIO
# Description:       Enable Garage Door GPIO
### END INIT INFO

PATH=/sbin:/bin:/usr/sbin:/usr/bin
DESC="Enable GPIO for the Garage Door Module"
NAME="garage-door-gpio"
SCRIPTNAME=/etc/init.d/$NAME

. /lib/lsb/init-functions

case "$1" in
    start)
        log_daemon_msg "Enabling Garage Door GPIO"
        success=0
        if [ ! -e /sys/class/gpio/gpio20 ] ; then
                echo 20 > /sys/class/gpio/export
                success=$?
                echo out > /sys/class/gpio/gpio20/direction
                echo 1 > /sys/class/gpio/gpio20/active_low
                echo 0 > /sys/class/gpio/gpio20/value
        if

        if [ ! -e /sys/class/gpio/gpio21 ] ; then
                echo 21 > /sys/class/gpio/export
                success=$?
                echo in > /sys/class/gpio/gpio21/direction
        fi

        log_end_msg $success
        ;;
    *)
        echo "Usage: $SCRIPTNAME {start}" >&2
        exit 1
        ;;
esac

exit 0

toshibochan, 2017-09-16:
wall button stop working because you probably you wired wrong to relay.

brettstah, 2017-09-16:
I'm still trying to locate the proper logs, but here are the other files (I've added .txt to them for uploading purposes) - note that the garage door opens as soon as the init.d script runs, before homebridge runs, so I think the issue isn't with homebridge per se:
[config.json.txt](https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor/files/1308319/config.json.txt)
[garage-door-gpio.txt](https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor/files/1308320/garage-door-gpio.txt)

brettstah, 2017-09-16:
OK, I think I figured out how to grab the boot up log data:
[journalctl.txt](https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor/files/1308326/journalctl.txt)

brettstah, 2017-09-16:
Toshibochan's script has these two additional "echo" commands that mine doesn't have:

echo 1 > /sys/class/gpio/gpio20/active_low
echo 0 > /sys/class/gpio/gpio20/value

Would those be relevant to the problems I've having?

brettstah, 2017-09-16:
Ok, so great success!  I added the two extra echo lines (changed 20 to 23), and once I restarted the Pi, the garage door did NOT open, and the wall button remained active/functional!  After homebridge started, the garage door status is correctly determined (my reed switch is using GPIO 18), and opening and closing the garage door works via homekit actions.  

Thanks for the help, folks!

closed 2017-09-16 by brettstah

Shareon01, 2018-11-01:
Hi Brettstah,
Im facing the same issue with the relay being triggered during boot.
Do you remember where exactly you added the extra echo lines?
Best Regards, Henrik