benlamonica/homebridge-rasppi-gpio-garagedoor #37: start script not found!

bozzomasterflash, 2017-12-22:
Hi,
first of all, thank you for this nice job!
If I try to download the start script (https://raw.githubusercontent.com/benlamonica/homebridge-rasppi-gpio-garagedoor/master/scripts/garage-door-gpio) 404 not found appears.
Could you please verify?
thanks
Stefano

c0redevelopment, 2018-03-11:
In the instructions there are two dead links to scripts: 

a. Copy and edit this start script into your /etc/init.d directory.
where "start script" link is:
https://raw.githubusercontent.com/benlamonica/homebridge-rasppi-gpio-garagedoor/master/scripts/garage-door-gpio

and also another broken link from the instructions:

a. Copying the homebridge start script into your /etc/init.d directory.
where the "start script" link is:
https://raw.githubusercontent.com/benlamonica/homebridge-rasppi-gpio-garagedoor/master/scripts/homebridge

benlamonica, 2018-03-11:
Sorry about that! I started upgrading to systemd scripts and then forgot to update the README.md.

The systemd scripts are there now. I'll update the README.

benlamonica, 2018-03-11:
I need to republish to npm. The instructions in the master repo are correct. I've fixed the formatting for them to make it more clear.

Please take a look at:
https://github.com/benlamonica/homebridge-rasppi-gpio-garagedoor/blob/master/README.md

And I will repush to npm shortly.

closed 2018-03-11 by benlamonica