benlamonica/homebridge-rasppi-gpio-garagedoor #38: Is it possible to simulate a closedDoorSensor?

0neiro, 2018-01-03:
Hello,
at first I want to thank you for the great plugin. It works very good!

I'm using the openDoorSensor and not the closedDoorSensor. Therefore Siri does not give me a promptly feedback when I ask "her" to open my front garage door. This is logical, since Siri (and accordingly homebridge) is waiting for the response from my openDoorSensor (the reed closes the circuitry when the door is fully opened). I could switch the sensor setup to a closed door sensor, but then I would have the same experience when I want to close the door. Just for the info: my Door needs 20 sec to open/close.

So I want to have a fast feedback from Siri (without waiting times) and without the need to install closed AND opend sensors. And here is my question:
Would it be possible to add a function to simulate a missing doorsensor? For example:
I ask Siri to open the garage door, your plugin/homebridge will open my door and right after this it will tell Siri, that the door is opened (without to wait for the openDoorSensor). So it will simulate a missing closedDoorSensor. If my openDoorSensor would not trigger after ~25 sec, your plugin could revoke the open status and correct it to closed.

I dont know if this would be possible, but I think with such an option, all my openings and closings will get a nice and fast response in homekit/siri.
Best regards!

ukypayne, 2019-11-16:
Why not just use the same sensor and invert the logic? You aren't really determining if the door is closed, but it would tell you if it is NOT open.

This isn't a great option since you don't have a confirmation it is actually closed, which is why a closed sensor install would have been preferred.

mikewiller314, 2023-02-13:
Thank you for reaching out and for your positive feedback on the Shortpump quick service plugin. I understand that you would like to have a faster response from Siri when you ask it to open your front garage door.As for your suggestion of simulating a missing door sensor, it is possible to add such a feature, but it would require some programming and technical knowledge to implement. If you are familiar with coding and are comfortable making modifications to the [Shortpump quick](https://garagedoorrepairshortpump.com/garage-door-replacement/) service plugin, you could try to implement this feature yourself. Alternatively, you could reach out to the developer of the Shortpump quick service plugin for assistance, or consider hiring a professional to help you make the necessary modifications.Please keep in mind that this type of customization may also have potential security and stability implications, so make sure to thoroughly test the changes before using it in a production environment.I hope this information helps, and if you have any further questions, please feel free to ask!"

closed 2026-11-13 by 0neiro