benlamonica/homebridge-rasppi-gpio-garagedoor #47: Trouble with closed and open sensor

NajibNour, 2018-12-27:
Hello, i recently tried to add a second sensor to my config i cannot seem to get it to work i belive its only using one sensor and not 2. this was my original config.
`    "accessories": [{
        "accessory": "RaspPiGPIOGarageDoor",
        "name": "Garage Door",
        "doorSwitchPin": 5,
        "doorSwitchPressTimeInMs": 1500,
        "doorSwitchValue": 1,
        "closedDoorSensorPin": 20,
        "closedDoorSensorValue": 1,
        "doorPollInMs": 4000,
        "doorOpensInSeconds": 12
    }],`

i have now switched it over too

`    "accessories": [{
        "accessory": "RaspPiGPIOGarageDoor",
        "name": "Garage Door",
        "doorSwitchPin": 5,
        "doorSwitchPressTimeInMs": 1000,
        "doorSwitchValue": 1,
        "closedDoorSensorPin": 24,
        "closedDoorSensorValue": 1,
        "openDoorSensorPin": 25,
        "openDoorSensorValue": 1,
        "doorPollInMs": 4000,
        "doorOpensInSeconds": 13
    }],`

the service runs fine but i feel like its only working on the closed door sensor when there is contact it says its closed but when i remove the contact it says its open regardless of the open door sensor.

iondarie, 2019-01-20:
i have the same problem but mine ignores the close sensor

iondarie, 2019-02-08:
have you found a solution to this?