benlamonica/homebridge-rasppi-gpio-garagedoor #49: Homebridge not starting

jabba243, 2019-11-10:
Anyone having issues with Geek Slant Smart Garage since updating to iOS 13? My setup was working fine until 13 then iPhone would respond  by stating “you have more than one garage door. Ask me again with a specific room or device name”. This is after I’ve asked for Siri to open/close right garage door. I’ve rebooted rpi and deleted the bridge from my phone and now can’t find it again. Now I get this error. 

[image]

skirkpatrick1979, 2020-01-09:
I am having this same issue.  Did you ever get an answer?  If I remove the closed sensor setting in my config.json, homebridge will run fine.  As soon as I place the setting back in the file, I get teh same error as above.

jabba243, 2020-01-09:
Not really. I ended up starting again with fresh install/setup. Fixed the issue in the debug but not the error where Siri was asking for more info on which door to open. Then I noticed that what my Apple Watch was hearing was different to what I was saying. Example if I said “open right garage door” Siri would hear “open a right garage door” thenI would get the response of “you have more than one garage door....”.  I have figured out how to fix that. Seems a Siri or watch issue more than the geek slant software.

jabba243, 2020-01-09:
* I haven’t figured out how to fix that.

toshibochan, 2020-01-09:
Now I’m using this! Take look
https://github.com/dubocr/homebridge-gpio-device

skirkpatrick1979, 2020-01-10:
Thanks.  I guess this isn't supported any more.  I will move to your suggested add on

albliu, 2020-06-03:
got this working yesterday. for me, i had to make sure homebridge calls the garage-door-gpio start script on start up. "see the step 3 in readme.md""

I had to manually edit the exisiting homebridge.service file though, instead of copying it over from the project, because i was running raspian set up from https://github.com/homebridge/homebridge-raspbian-image/wiki/Getting-Started, and they had a homebridge.service already

benlamonica, 2020-06-13:
Sorry, I got a new garage door that doesn't have a simple mechanism, and just don't have time support the project.

closed 2020-06-13 by benlamonica