benlamonica/homebridge-rasppi-gpio-garagedoor #52: Working on homebridge 1.0.x and/or newest npm/node.js-Version

Stefan-83, 2020-04-29:
Hi,
i'm currently using this plugin on homebridge 0.4.53 , Node.js Version  v12.16.2 and NPM v6.9.0.
Before I update to the most current versions: Are there any experience, that the plugin is working on that?

Thanks,
Stefan

benlamonica, 2020-04-29:
Unfortunately no. I ended up getting a new garage door a while ago and haven’t had the time to get it working again as the button is no longer a simple switch, but the signal needs to be encoded. So I’m not sure what would happen. 

-Ben L

Stefan-83, 2020-04-29:
Hm, I see..
This means, if something would break after updating the homebridge, you wouldn't be able to help with an updated version of this plugin, right?

Stefan-83, 2020-04-29:
Update: I did the update homebridge 1.0.3 and it works :)
I didn't update Node.js or NPM (still on Node.js Version v12.16.2 and NPM v6.9.0)