benlamonica/homebridge-rasppi-gpio-garagedoor #53: cant find GPIO directory?

Lummelyyy, 2020-07-01:
hi,
currently trying to get this plugin to work but give's me this error. i am fairly new to raspberry's so i must be doing something wrong here. anybody any idea's
Error: ENOENT: no such file or directory, open '/sys/class/gpio/gpio24/value'
    at Object.openSync (fs.js:443:3)
    at Object.readFileSync (fs.js:343:35)
    at RaspPiGPIOGarageDoorAccessory.readPin (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:149:26)
    at RaspPiGPIOGarageDoorAccessory.isClosed (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:158:21)
    at RaspPiGPIOGarageDoorAccessory.initService (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:124:25)
    at new RaspPiGPIOGarageDoorAccessory (/usr/local/lib/node_modules/homebridge-rasppi-gpio-garagedoor/index.js:66:8)
    at config.accessories.forEach (/usr/local/lib/node_modules/homebridge/src/server.ts:357:50)
    at Array.forEach (<anonymous>)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/src/server.ts:330:29)
    at Server.start (/usr/local/lib/node_modules/homebridge/src/server.ts:156:12)

i use 
node.js v10.19.0
NPM v6.14.5
if that helps

Thanks,
Michel

toshibochan, 2020-07-01:
Use homebridge-gpio-device

toshibochan, 2020-07-01:
If homebridge-gpio-device give you error use homebridge-gpio-device-orpi

toshibochan, 2020-07-01:
This is my config exemple for homebridge-gpio-device

"accessory": "GPIODevice",
                "type": "GarageDoorOpener",
                "name": "Gate",
                "pin": 4,
                "inverted": "false",
                "pulseDuration": 200,
                "invertedInputs": "false",
                "openSensorPin": 24

Lummelyyy, 2020-07-01:
Thank you!

I still got errors with homebridge-gpio-device
But with Homebridge-gpio-device-orpi it seems to be fixed!

closed 2020-07-01 by Lummelyyy