benlamonica/homebridge-rasppi-gpio-garagedoor #55: echo [pin#] /sys/class/GPIO/export no longer works

ozjc, 2024-04-21:
I've been happily using homebridge-rasppi-gpio-garagedoor for a few years now, to monitor and control my two garage doors. Thanks for the really useful plug-in.

Unfortunately it looks like the latest update to Raspi OS has finally broken the /sys/class/gpio/ method for setting & getting GPIO pin states.
$ uname -a
$ Linux raspi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
$ sudo echo 17 > /sys/class/gpio/export
echo: write error: Invalid argument

I *think* the OS kernel change to 6.6.20 is what's broken things. My research suggests that there's a need to change to using pinctrl or to replace RPi.GPIO with rpi-lgpio. The second is a bit of a guess, based on the assumption that the sys/class/gpio method relies on RPi.GPIO. But I'm a rank amateur so I may well be wrong on both fronts.

Hope you're still watching over this!

benlamonica, 2024-04-21:
Unfortunately I got a new garage door opener that this no longer works on, so I have no way of testing. If you submit a pull request I’ll merge it and push the changes though to npm.

ozjc, 2024-05-05:
I think I see what's happened. I'm installing your plugin via the Homebridge GUI, which installs it using npm. Even though Homebridge thinks it's installing v1.0.10 of your plugin, that appears to be a much older version than the one in this git repository. The one here uses rpio, whereas the one installed by Homebridge installs an old one using rpm-gpio (and the now deprecated-with-extreme-prejudice /sys/class/gpio method). Any chance you could update the rpm version to match the one here? That would at least mean Homebridge is actually installing your latest version.

I hope this isn't complete rubbish!

benlamonica, 2024-05-05:
Yes. I can do that. I will look into it tomorrow-Ben L

mikegabriel, 2024-05-08:
@ozjc thanks for reporting this, I just stumbled into this issue today when reinstalling a clean RPi OS Lite.  

That being said, it looks like this package is now broken with the latest homebridge using nodejs v20+ due to some dependencies on rpio that are fixed in newer versions than this package depends on.

Can you tell me what version of homebridge / node you are using if this is still working for you?

ozjc, 2024-05-09:
It’s not working for me either. But it appears to me there are issues with the JavaScript. For example, there’s an attempt to read a GPIO pin before it’s been opened (by the init function). I don’t have much time, but I’m working backwards through the errors. I may well run into the issues you describe but I didn’t in my first iteration, just bumped into another sequencing issue (I think). I’ll let you know what I end up with. Being a very amateur and incompetent coder I can’t guarantee any silver bullet!I’ll let you know what node/hb versions I’m using once I’m back at my desk. Cheers,John Cunninghamm: +61 419 120 844

mikegabriel, 2024-05-09:
@ozjc You're correct, I think it's line 128 which attempts to read the closed sensor pin before opening it.  In my setup I have a closed sensor but no open sensor but I am still going through the code and comparing to the old version that I was running without any issues, going to try and get a working copy restored today.

I touched on it in my previous comment but it looks line one of the most recent commits changed lines 114 and 118 in an unintended way.  This is assuming you do not have at least one of the open/closed sensors in your config.

```
  hasOpenSensor : function() {
    return this.openDoorSensorPin !== null;
  },

  hasClosedSensor : function() {
    return this.closedDoorSensorPin !== null;
  },
```

The commit changed `!=` to `!==` which compares `Type` and now these functions will always return `true` since you either set a value for these pins or didn't.  But if you didn't, the pin is set as `undefined` and not null.  There's 2 probable fixes.

1. Revert back to original so that the comparison doesn't factor in type: 
* `return this.openDoorSensorPin != null;`
3. Change null to undefined, I think this is valid and what the author intended since the json config would never allow for null values and the pins are initialized in the script either way: 
*  `return this.openDoorSensorPin !== undefined;`

This got my plugin to boot, previously it was trying to read off the closed pin (which I have) before opening it.  Opening it does happen in the initService function on line 149 but was

mikegabriel, 2024-05-09:
I believe I got everything working in my setup (has closed sensor but no open sensor).

I've staged everything so it could be incorporated via a pull request or just so you can look through it: https://github.com/mikegabriel/homebridge-rasppi-gpio-garagedoor/commits/dependency-update-and-fixes/

ozjc, 2024-05-09:
That’s great - I’ll look at it this weekend as well. Like you, I have only a closed sensor. I also have a second door. John Cunninghamm: +61 419 120 844

ozjc, 2024-05-11:
Thanks Mike - all working in my setup, with a couple of issues/workarounds.
$ uname -a
Linux acunninghome 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
`$ hb-shell
$ homebridge --version
1.8.1
$ node --version
v20.13.1
$ npm --version
10.5.2`

I couldn't install from your repository using rpm, so I installed rpio via rpm and then Ben's version of the plugin via the Homebridge GUI. Then I made your changes manually to index.js and package.json. I also needed to change the switch value in the config file, but from 0 to 1 in my case!

Only remaining issue is the error in the lines below (from the very end of the startup log) which doesn't seem to affect functionality:
`[11/05/2024, 3:05:49 pm] Homebridge v1.8.1 (HAP v0.12.0) (Left Garage Door) is running on port 37010.
[11/05/2024, 3:05:50 pm] [Right Garage Door] GarageDoor is CLOSED (1)
[11/05/2024, 3:05:50 pm] [homebridge-rasppi-gpio-garagedoor] This plugin generated a warning from the characteristic 'Target Door State': characteristic value expected valid finite number and received "undefined" (undefined). See https://homebridge.io/w/JtMGR for more info.
[11/05/2024, 3:05:51 pm] [Left Garage Door] GarageDoor is CLOSED (1)
[11/05/2024, 3:05:51 pm] [homebridge-rasppi-gpio-garagedoor] This plugin generated a warning from the characteristic 'Target Door State': characteristic value expected valid finite number and received "undefined" (undefined). See https://homebridge.io/w/JtMGR for more info.`

mikegabriel, 2024-05-21:
@ozjc I tried to fix that target door state error briefly but didn't have any luck and figured it wasn't hurting anything.  The initial value is empty and that's what causes the issue, it gets set once it polls the pins and becomes happy.  I figured it wasn't hurting anything so I left it.

For anyone else that might find this and need help, this is how I installed my version of the plugin.

```
hb-shell
npm --prefix "/var/lib/homebridge" install --save https://github.com/mikegabriel/homebridge-rasppi-gpio-garagedoor/tarball/master
```

medo145, 2024-06-06:
@mikegabriel Thanks for taking the time to make the required changes to revive this plugin. I've tried many others after this one stopped working, but all of them had some kind of issue/bug that prevented me from using it.

The command you left in your last post did not work for me, but changing the URL to point to your changes worked. 

```
hb-shell
npm --prefix "/var/lib/homebridge" install --save https://github.com/mikegabriel/homebridge-rasppi-gpio-garagedoor/tarball/dependency-update-and-fixes
```

I also had to change doorSwitchValue to 0 from 1 to prevent it from opening the door on reboot.

Thanks again!!

Dillinja2, 2024-06-16:
Want to say thanks to @mikegabriel and folks helping keep this project alive.

I'm currently having an issue with the systemd startup scripts, specifically a compatibility with homebridge.service and the modern way hb-service works on startup. Can someone share their current homebridge.service startup script that starts the GPIO exports but is compatible with hb-service? Thanks!

mikegabriel, 2024-06-19:
@Dillinja2 I don't want to stray off topic too much but here's mine from a fresh install of everything, RPi OS Lite and Homebridge.

```
$ sudo cat /lib/systemd/system/homebridge.service
[Unit]
Description=Homebridge
Wants=network-online.target
After=syslog.target network-online.target

[Service]
Type=simple
#User=homebridge
User=root
SupplementaryGroups=homebridge
PermissionsStartOnly=true
WorkingDirectory=/var/lib/homebridge
ExecStartPre=-/bin/run-parts /etc/hb-service/homebridge/prestart.d
ExecStart=
ExecStart=/opt/homebridge/start.sh --allow-root
Environment="UIX_CAN_SHUTDOWN_RESTART_HOST=1"
Restart=always
RestartSec=3
KillMode=process
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_NET_RAW CAP_NET_ADMIN

[Install]
WantedBy=multi-user.target
```